lumapu/ahoy #171: Active Power Limit Control Type  ändert sich über MQTT

HFarryranz, 2022-08-18:
0.5.14
ich stelle im Setup
- Active Power Limit Control Type auf 'relative in percent non persistent'
- Active Power  auf 100

in der Visualisierung wir auch 100% angezeigt.
Schreibe ich nun in mqtt.0.inverter.devcontrol.0.11  den Wert 80 
wird  Active Power Limit Control Type auf 'absolute in Watt non persistent' und  die Anzeige wird zu 100W

wäre auch schön,
- wenn man den Wert im Setup einträgt, dass auch auf MQTT-Seite der Wert aktualisiert wird.
- Active Power Limit Control Type  über MQTT verändert werden kann

aschiffler, 2022-08-18:
Hi,
das stimmt. Das ist so. Der Zahlenwert der im Setup steht wird NICHT durch ein Setzen seitens MQTT verändert.
Die Intention ist so gedacht. Setup -> "Das will ich einmalig oder eben beim Setup einstellen" "Das Setup ist keine Live-Feedback-Anzeige".

Dieser Punkt:
> wenn man den Wert im Setup einträgt, dass auch auf MQTT-Seite der Wert aktualisiert wird.
wird nie richtig robust funktionieren solange es keine Möglichkeit gibt das aktuelle Limit aus dem WR abzufragen. Daher gibt es diesen "Output" nicht.

Dieser Punkt:
> Active Power Limit Control Type über MQTT verändert werden kann
ist bereits implementiert. Man muss das Topic entsprechend erweitern. Siehe die knappe Anleitung hier unter der Rubrik MQTT
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/User_Manual.md#active-power-limit-via-mqtt

aschiffler, 2022-08-18:
Hi,

danke für das Testen / Feedback.
Das Setup ist keine Live-Anzeige. Daher werden hier die Werte die ggf. per MQTT geändert werden nicht durchgängig aktualisiert.

Zu deinen Punkten.
> wenn man den Wert im Setup einträgt, dass auch auf MQTT-Seite der Wert aktualisiert wird.
Das wird nie robust funktionieren solange es nicht möglich ist das aktuelle Limit aus dem WR auszulesen. Daher gibt es diesen Output nicht.
> Active Power Limit Control Type über MQTT verändert werden kann
Das ist bereits implementiert. Siehe die knappe Anleitung hier unter der Rubrik MQTT
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/User_Manual.md#active-power-limit-via-mqtt

closed 2022-08-18 by aschiffler