lumapu/ahoy #178: DevInfo Cmd Parser für 0x00 InverterDevInform_Simple bis 0x05 SystemConfigPara

stefan123t, 2022-08-19:
Alle zum MainCmd 0x15 REQ_ARW_DAT_ALL bzw. dem. Device Information gehörenden SubCmd Kommandos implementieren:

* [ ]  0x00 InverterDevInform_Simple
* [x]  0x01 InverterDevInform_All
* [ ]  0x02 GridOnProFilePara
* [ ]  0x03 HardWareConfig
* [ ]  0x04 SimpleCalibrationPara
* [ ]  0x05 SystemConfigPara

Kommandos bzw Parser analog zu 0x01 InverterDevInform_All implementieren und Daten in WF Struktur an/ablegen.

aschiffler, 2022-08-19:
@stefan123t 
Hi, ich würde das noch etwas zurück stellen.
Wenn ich in unser Backlog #142 schaue steht dort "AlarmData abrufen, parsen und Anzeigen". Bzw. vorher "Scheduler"

aschiffler, 2022-08-21:
Hi,
From SubCmd 0x5 now the parser is implemented. Only Bytes 2 and 3 are use (acht powerlimit)
Any SubCmd is now possible but parsers still missing. Esp. Parser for Alarm Data.