lumapu/ahoy #182: MQTT connect with user + password not possible; MQTT Password length only 31 chars possible

Chrischi-, 2022-08-19:
MQTT connect with user + password not possible

Mosquitto MQTT Broker Log:
```
1660944902: New connection from 172.16.0.29:55936 on port 1883.
1660944902: Client <unknown> disconnected, not authorised.
```

MQTT Password length only 31 chars possible
Everything after 31 characters is truncated and not saved.

I have tested all versions from 5.5 to 5.14 without success. I had not yet time to look for the issue in the source.

stefan123t, 2022-08-19:
Most likely a one off limitation as we are writing only 32 chars into EEPROM we may miss the \0 String ending. We had this before for WiFi Password too

Chrischi-, 2022-08-20:
Brief additional info:
Login is not possible even with shorter password. 
However, a MQTT listener **without** authentication works without problems.

Tested up to 5.15

aschiffler, 2022-09-04:
Will leave this open for the next release >0.5.16

Chrischi-, 2022-09-04:
Maybe it was me, but with 0.5.16 I have no more problems logging in. 

However, the issue with the password length still exists.

aschiffler, 2022-09-04:
Close this bc. password length of 31 characters shall be sufficient.

closed 2022-09-04 by aschiffler

Chrischi-, 2022-09-04:
@aschiffler 
OK, then I would also include the maximum password length somewhere in the documentation. Otherwise, this issue will surely repeat itself at some point.