lumapu/ahoy #183: Settings not stored

Fellhund, 2022-08-20:
AHOY,

I now also have the problems, also in the current version,
that values ​​are not saved.

Within Section Inverter, Address* and Name* are not saved.
Tried various browsers, various sequences, SAVE with reboot, without
reboot...

Device name and WLAN settings are retained.

What's the matter?

tastendruecker123, 2022-08-20:
Has it accepted any settings at all in the inverter section or is it completely blank now? Have you tried erasing the settings yet?

roku133, 2022-08-20:
v0.5.14 in main branch shows this problem (see issue #162),
currently, the development release has to be installed to solve this issue.

Solar320, 2022-08-20:
> AHOY,
> 
> I now also have the problems, also in the current version, that values ​​are not saved.
> 
> Within Section Inverter, Address* and Name* are not saved. Tried various browsers, various sequences, SAVE with reboot, without reboot...
> 
> Device name and WLAN settings are retained.
> 
> What's the matter?

I have version 0.5.15 installed for ESP8266. Now it work but I don't know what the reason for it. 
I have done the following (to many at once ) things: 
-Erase the complete Flash memory of the ESP8266 with an ESP flasher Software.
-Change the // number of configurable inverters to #define MAX_NUM_INVERTERS       1
-Flash the latest V0.5.15
-insert all data to the Inverter and W-Lan connection
-set the "Amplifier power Level " to min!! Otherwise no or an really bad or non connection to the Inverter is possible.
  Í have only 5 meters distance to the Inverter, may be it's to loud for the receiver of the Inverter?
I remove the power of the ESP8266 for 10 seconds and repower the ESP and it remember all the settings I put in, w-lan and Inverter.
Also there is an error message inside PlattformIO: .pio\libdeps\esp8266-release\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\esp8266-release\PubSubClient\src\PubSubClient.cpp:523:16: warning: comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]
  523 |     return (rc == expectedLength);

roku133, 2022-08-20:
v0.5.15 now available in main branch

stefan123t, 2022-08-20:
@Fellhund and @Solar320 the main problem with #162 has been fixed in 0.5.15 please retry with todays release. I will close this as duplicate of #162. Please reopen if you have been able to reproduce it in 0.5.15. Thanks

Regarding the PubSubClient warning this can be ignored and should be fixed upstream. We have mentioned it in the https://github.com/knolleary/pubsubclient/issues/895

closed 2022-08-21 by aschiffler

Solar320, 2022-08-21:
The error is still present. If you configure the Inverters, Inverter 0 is saved. But Inverter 1 has allready the Serial "484d2d3830300000" and the Power Limit "65535" inserted. If you delete this and save and restart the ESP8266, the values are not deleted and are inserted again in the Setup. The consequence is that the HTML Surface shows 2 Inverters although only on was configured.  I will try to change the values to other values than "nothing" like1234567 and 16000 and will look what happen. If this values also not saved this shows that there is something wrong. Since I have only one Inverter my fast solution was to change the max supported Inverters from 3 to 1 to prevent to showing 2 Inverters at the visualization HTML site.

Solar320, 2022-08-21:
Ok I get it, the problem is that this html surface will only change anything when you not insert capitals and numbers. If you don't insert anything in the fields (or delete the old input) , the HTML Surface will do nothing and use the old values. But the Surface should delete the memory if I delete alle numbers and capitals in the input fields.
I can confirm that an new Inverter will saved, if you change anything but there is still the problem that you can't save an empty inputfield.  I can't confirm if an second Inverter will work because I have only one.

Solar320, 2022-08-21:
I found an quickfix for this problem. If you want to let the support in the source code at 3 Inverters and have currently  only one Inverter installed, then you can set the second Inverter input fields all to "0".  If you insert "0" instead of deleting all the input of the input fields, it will be interprated as "Nothing" or no Value. If you then save and restart, all the inputfields of the Inverter 1 will be clean also in the vizualization only one Inverter will be showed.