lumapu/ahoy #184: Various bugs in current development version, mostly related to /setup

tastendruecker123, 2022-08-20:
-JS function checkSerial() in setup.html is not present and leads to JS errors in browser
-Bad MQTT broker hostname in settings leads to mqtt.h making ~1000 connection attempts per second to the non-working host
-Buffer arrays not being initialized properly in various places (search code for '{0}')
-Inverter address cannot be shortened once configured. Attempting to change it from 3823398298 to 382 results in 3820000000 being stored.
-Several parameters cannot be deleted or are not being stored properly when they're empty (inverter name, active power limit and all MQTT parameters for example).

DanielR92, 2022-08-20:
I make a PR for the first line, I check that with web.browser error and can confirm it. The function is not exist. - Thanks

tastendruecker123, 2022-08-20:
Lovely. This should take care of bug #1.

lumapu, 2022-08-21:
what do you mean by 'bad broker host name'? A non existing host or an empty field?

tastendruecker123, 2022-08-21:
A host that doesn't resolve like "test".

lumapu, 2022-08-30:
> Buffer arrays not being initialized properly in various places (search code for '{0}')
done.

closed 2022-09-04 by aschiffler

stefan123t, 2022-09-04:
1. checkSerial was fixed
2. MQTT broker name flood control has been implemented as #194
3. buffer arrays have been implemented 
4. overwriting old values in setup ?
5. delete or store empty values in setup ?

@aschiffler is 4 and 5 fixed in the EEPROM class too, ie can we overwrite parts of the settings now ?