lumapu/ahoy #185: Auswertung und Diagramm Integration

DennisXK, 2022-08-20:
Hallo, wäre es möglich eine Auswertung in Form einem Diagramm von der Tagesleistung, denn erzeugen kwh pro Tag zu integrieren? würde das der Controller noch packen? Denn nicht jeder hat einen IOBroker wo er Garfana etc. am laufen hat.

DanielR92, 2022-08-20:
Hallo Dennis, theoretisch wäre dies möglich, nur müssen wir schauen inwieweit wir das intigrieren können. Das Problem bei der Auswertung ist das man die Daten zwischen speichern müsste um ein Diagramm generieren zu können und das Thema Speicher ist hier problematisch. ESP8266 hat nicht genug speicher für eine längere Zeit.

Man müsste es mal "berechnen" wie lange man die Daten (Auflösung und Zeitliche länge) speichern könne. 
Daher bleibt dieser Punkt vorerst noch ein Feature.

tastendruecker123, 2022-08-21:
Möglich wäre es, Daten an Thingspeak zu liefern. Mit einem kostenlosen Account dort kann man z.B. [sowas](https://thingspeak.com/channels/1730810) machen. Ebenso ist es möglich, die gespeicherten Daten per JSON wieder zu lesen, also könnte man diese JSON Daten von Thingspeak im Web-Interface von ahoy in einem Diagramm anzeigen lassen. Der Entwicklungsaufwand wäre nicht sehr hoch da Thingspeak eine einfache API bietet (HTTP Post, MQTT usw.) und man müsste lokal nichts speichern.

tavoc, 2022-08-21:
Guten Morgen,

eine Verständnisfrage von mir. Ich habe einem Freund bereits ein funktionierendes Setup gebaut (mit esp8266) und ich warte noch auf meine Solarmodule. Nach meinem kurzen Blick liefert der Inverter kumuliert die Werte für die Gesamteinspeisung. Verliert er diese, sofern er spannungslos ist.

Wie wäre es, wenn man die Übersichtstabelle (Visualisierung grüner Bereich) stündlich/täglich in eine txt Datei schreibt, beispielsweise per SMB/FTP auf seine Fritzbox. Dann würde man sich den Smarthomeserver sparen und es reicht für eine minimale selbstauswertung.
Geht ein derartiger Export?

fila612, 2022-08-21:
Was meinst du genau mit Gesamteinspeisung?

Der WR liefert die Erträge pro Tag und gesamthaft in kWh.
Die aktuelle Leistung in W wird für AC und DC dargestellt.

Aus meiner Sicht reichen die Werte, aber ich bin nicht sicher was du brauchst.

tavoc, 2022-08-21:
Ich dachte an einen täglichen Export der genau diese Werte in ein externes Verzeichnis schreibt. 

Z.b.
2022_08_21.txt

Dann kann man sich aus diesen Daten z.b. per Excel oder jedem anderen Programm selbst seine Diagramme bauen. 

Ich selbst nutze fhem/mqtt und werde es wohl nicht brauchen. Bei meinem Kumpel läuft jedoch bis auf die Fritzbox kein Gerät 24/7. das wäre dann eine Möglichkeit die Daten extern zu persistieren.
Ich weiss aber nicht ob der ESP sowas überhaupt kann.

Und man braucht sich keine Gedanken über Schreibvorgänge im flash machen oder Speicherplatz auf dem ESP selbst

DennisXK, 2022-08-21:
Alles gute Ideen, ich selbst nutze auch mqtt und iobroker/Grafana um das auswerten zu können was die Anlage produziert hat. Ich denk aber die meisten habe das nicht, würden aber gerne sehen wollen wieviel sie von Tag zu Tag produziert haben.

Jeden Tag eine txt file zu erstellen ist bestimmt aufwendig wenn man im Excel die Auswertung für das ganze Jahr machen möchte.

man könnte doch auch ein SD Modul einbauen.

fila612, 2022-08-21:
der Speicher im ESP ist ja begrenzt, die Entscheidung die man m.M.n. zu Beginn treffen sollte ist, ob es Kapazitäten für solche Daten auf dem ESP gibt oder nicht.
Falls (begrenzten oder definierten) Speicher dafür genutzt werden kann (ich meiner ein Wemos D1 hätte 4MB Flash), dann käme ja evtl. auch eine CSV Lösung in Betracht.
ich habe keine Ahnung ob diese hier etwas wäre, aber auf die schnelle mal gesucht:
https://tttapa.github.io/ESP8266/Chap16%20-%20Data%20Logging.html

stefan123t, 2022-08-21:
@DennisXK SD Card Reader support gibt es bereits in PR https://github.com/stefan123t/ahoy/pull/1 von Gerald R aus dem Forum. Ich weiss nur nicht ob sich der aktuell mergen lässt.

FTP/SMB Protokoll dürfte etwas knifflig werden es sei denn jemand kennt eine gute Bibliothek für so etwas. Ich habe auch schon an WebDAV gedacht das wird idR von Fritz!Box und Nextcloud angeboten aber ich wüsste nicht wie man damit ein POST / UPDATE zum Append einer einzelnen Zeile an eine Datei umsetzt. 

Ich selber habe in der Vergangenheit das SensorLogger Plugin in Nextcloud genutzt aber das ist evtl nicht so gut maintained bzw läuft auf den aktuellen Versionen NextCloud >24 nicht sofort.

tavoc, 2022-08-21:
Habe das hier für FTP gefunden. Ein monatliches CSV

https://forum.arduino.cc/t/esp8266-smb-dateizugriff-moglich-sowie-frage-zur-datenvisualisierung/689277/4

if (ftp_write())Serial.print(F("FTP ok")); // Aufruf wann erforderlich

//================ FTP ==============================
// -------------- ftp recive ---------------------
byte eRcv() {
  byte respCode;
  byte thisByte;

  while (!ftpClient.available()) delay(1);
  respCode = ftpClient.peek();
  byte i = 0;
  while (ftpClient.available()) {
    thisByte = ftpClient.read();
    if (i < 200) {
      buffer[i] = thisByte;
      i++;
      buffer[i] = 0;
    }
  }
  Serial.print(respCode); Serial.print("  "); Serial.println(buffer);

  if (respCode >= '4') {
    efail();
    return 0;
  }
  return 1;

}
//----------------- ftp client fehler ----------------
void efail() {
  byte thisByte = 0;

  ftpClient.println(F("QUIT"));
  while (!ftpClient.available()) delay(1);
  while (ftpClient.available()) {
    thisByte = ftpClient.read();
    Serial.write(thisByte);
  }
  ftpClient.stop();
  Serial.println(F("Abbruch FTP getrennt"));

}

//--------------------- FTP daten schreiben-------------

byte ftp_write() {
  char file [20];
  sprintf(file, "%02d%02dwet.csv", jahr - 2000, monat);

  char DatumZeit[40];
  sprintf(DatumZeit, "%02d.%02d.%04d %02d:%02d:%02d",
          lasttag, lastmonat, lastjahr, lasthour, lastmin, lastsec);

  //  sprintf(DatumZeit, "%04d-%02d-%02d %02d:%02d:%02d",
  //          jahr, monat, tag, stunden, minuten, sekunden);

  Serial.println (file);
  Serial.println(DatumZeit);

  if (ftpClient.connect(gateway, 21)) {
    Serial.println(F("FTP Server verbunden"));
  }
  else {
    Serial.println(F("FTP Verbindung nicht möglich"));
    return 0;
  }

  if (!eRcv()) return 0;

  Serial.println(F("send user"));
  ftpClient.println(F("USER xxxxx"));// xxx Benutzer
  if (!eRcv()) return 0;

  Serial.println(F("Send password"));
  ftpClient.println(F("PASS yyyyy"));// yyy password
  if (!eRcv()) return 0;

  Serial.println(F("send SYST"));
  ftpClient.println(F("SYST"));
  if (!eRcv()) return 0;

  Serial.println(F("Verzeichniss wechseln"));
  ftpClient.println(F("CWD STICK"));
  if (!eRcv()) return 0;

  Serial.println(F("CWD daten"));
  ftpClient.println(F("CWD daten"));
  if (!eRcv()) return 0;

  Serial.println(F("pwd"));
  ftpClient.println(F("pwd"));
  if (!eRcv()) return 0;

  Serial.println(F("TYPE I"));
  ftpClient.println(F("TYPE I"));
  if (!eRcv()) return 0;

  Serial.println(F("PASV"));
  ftpClient.println(F("PASV"));
  if (!eRcv()) return 0;

  // Port auslesen steht nach 4 und 5 Komma
  char *tStr = strtok(buffer, "(,");
  int array_pasv[6];
  for (int i = 0; i < 6; i++) {
    tStr = strtok(NULL, "(,");
    array_pasv[i] = atoi(tStr);
    if (tStr == NULL) {
      Serial.println(F("Bad PASV Answer"));
    }
  }
  unsigned int hiport, loport;
  hiport = array_pasv[4] << 8;
  loport = array_pasv[5] & 255;
  hiport = hiport | loport;
  Serial.print(F("vom Server Port für Daten erhalten: ")); Serial.println(hiport);

  if (dClient.connect(gateway, hiport)) {
    Serial.println(F("FTP Dateserver verbunden"));
  }
  else {
    Serial.println(F("Daten-Verbindung nicht möglich"));
    return 0;
  }

  Serial.println(F("send Append"));
  ftpClient.print(F("APPE ")); ftpClient.println(file);

  if (!eRcv()) {
    dClient.stop();
    return 0;
  }
  String buf, buf1;
  buf += temp;
  buf += "; ";
  buf += hum;
  buf += "; ";
  buf += pres;
  buf += "; ";
  buf += VCC;
  buf.replace(".", ","); // punkt gegen Komma tauschen sonst wird das mit Excell nix

  buf1 = DatumZeit;
  buf1 += "; ";
  buf1 += buf;
  buf1 += ("\r\n");

  dClient.print(buf1);
  delay(1);
  dClient.stop();
  Serial.println(F("QUIT"));
  ftpClient.println(F("QUIT"));
  if (!eRcv()) return 0;
  ftpClient.stop();
  Serial.println(F("Verbindung normal beendet"));

  return 1;

}

DennisXK, 2022-08-21:
Also die Lösung die Daten an einen FTP Server der auf der Fritzbox läuft in eine File zu schreiben finde ich gut. 
Aber wie geht es dann weiter? 
Die Werte aus der file in ein Excelsheet zu integrieren finde ich nicht so toll. Sollte mehr Plug and Play sein, so dass es für User die nicht so den Plan von dem ganzen haben eine Visulasierung haben. 

Die beste Variante wäre wenn das Diagramm/e auf dem Dashboard der Ahoi dargestellt würde.

tavoc, 2022-08-21:
Ich könnte ein xls erstellen, mit vorgefertigtem datenimport. Man muss den Pfad anpassen. 

Man hätte also eine Pivottabelle und kann beliebige Diagramme/ Filter setzen. 
Ich schau mir das morgen mal an inwieweit das noch so geht.

tastendruecker123, 2022-08-21:
Um wieviele unterschiedliche Werte geht es, die gespeichert werden sollen?

DennisXK, 2022-08-21:
das könnte man ja festlegen, wenn es um die Performance geht. Aber 4 Werte sollte schon drin sein...
-Momentan Leistung
-Tagesleistung
-Temperatur 
-...

Daraus könnte man dann ein paar Diagramme erstellen.

tastendruecker123, 2022-08-21:
Sofern man keine riesige Menge an Werten braucht, halte ich Thingspeak für die wesentlich bessere Lösung. Kaum Entwicklungsaufwand, keine Schreibzyklen auf dem ESP für Zwischenspeicherung, Visualisierung gibt's dazu, kein lokaler Server nötig und das Ahoy Web-Interface könnte die Daten immer noch über JSON abrufen von TS um eine eigene Visualisierung im Interface anzuzeigen.

stefan123t, 2022-08-21:
Die folgenden Hinweise zum Design des LittleFS sollten wir in Betracht ziehen falls wir unsere Settings #164 oder wie hier ein lokales CSV File #185 mit den Daten der letzten Stunden oder besser bloß der ganzen Tage führen wollen:
https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md
Eventuell sind auch die unter Existing designs? 2. genannten Logging File Systeme wie JFFS, YAFFS oder eben das bekannte SPIFFS für unseren hier gewünschten Einsatzzweck (append an CSV Datei) besser geeignet ?

Vor allem die in folgendem LittleFS Issue genannten Probleme mit wear-leveling und bei den im ESP8266 eingesetzten SPI-NOR Flash Chips kritischen 10.000 block erase cycles (idR <<100.000) sollten wir ggf. unter Verwendung eines virtual flash devices mit der entsprechend passenden erase block size simulieren. 
https://github.com/littlefs-project/littlefs/issues/374
Wer kann denn die beim Wemos D1 mini und NodeMCU v2 bzw. den darauf verbauten ESP12E/F Modulen eingesetzten SPI-NOR Flash Typen identifzieren bzw. Angaben zur Erase Block Size machen ?

DennisXK, 2022-08-22:
@ tastendruecker123 ja hast recht Thingspeak wäre da eig schon perfekt. 8 Werte sollte auch jedem reichen und wenn man es dann noch in die Ahoy bekommt wäre das doch eine runde Sache.

lumapu, 2022-08-22:
@stefan123t SPI Flashes haben meist auch ein Blick erase was 4kB entspricht (Kommando 0xD8), Sonst gibt es noch sector erase und Chip erase.
Bei SPI (NOR) Flashes ist es auch so, dass man nur zuvor gelöschten Bereich beschreiben kann. Beim löschen wird alles auf 0xff geschrieben, d.h. bei eigentlichen schreiben werden nur noch die 0er gesetzt. Wenn man also eine Zeile hinzufügt müsste man sich merken, bis wo schön gelöscht wurde. Beim aktualisieren muss man einen 4k Block zwischenspeichern.

@all ich sehe den ESP als Datenquelle also Sensor und nicht als kleine smart Home Zentrale, daher kann die geforderte Funktionalität ein Derivat von den sein was wir als Ahoy entwickeln, aber aus meiner Sicht nicht Mainstream.

tastendruecker123, 2022-08-22:
Bei einem kostenlosen Account hat man maximal 4 Kanäle mit jeweils 8 Werten, die man nutzen kann, also insgesamt 32 Werte. Dazu hat man ein Limit von 3 Millionen messages pro Jahr. Bei voller Ausnutzung könnte man die 4 Channels also ca. alle 60 Sekunden aktualisieren. Bei nur einem Channel alle 15 Sekunden. Eventuell funktioniert es auch schon mit der jetzigen Version von ahoy wenn man den MQTT Broker von TS benutzt.

DennisXK, 2022-08-24:
Das heißt jetzt? Wird die Idee umgesetzt? Thingspeak wäre doch gut geeignet für die Umsetzung und 4 Diagramme mit 8 Werten hat man ja in der Regel nicht.  
Ich habe zb. Zwei Diagramme im Grafana.
-	Aktuelle Leistung, Wechselrichter Temperatur  "Linendiagramm"
-	Erzeugte kw/h (das man in Grafana  in Tage/Monate/Jahre filtern kann) "Balkendiagramm"

Chrischi-, 2022-08-25:
Mal eine Verständnisfrage. Wenn sämtliche Graphen doch bei ThingSpeak & Co. liegen, 
warum will man diese dann im Webinterface des ESP? 
Welchen Mehrwert hat das?

Warum nicht den ESP machen lassen, was er gut macht. Eine Datenquelle sein bzw. Gateway dafür spielen.

lumapu, 2022-08-25:
weil es sich über ein Webinterface wesentlich einfacher konfigurieren lässt sie über MQTT oä. 
Über die angezeigten Werte kann man abschätzen ob die Kommunikation soweit steht. Ich denke gerade bei einer Fehlersuche wenn in MQTT Broker nichts ankommt ist jeder dankbar irgendwo Anhaltspunkte zu finden woran es liegt.

Aus meiner Erfahrung schaue ich auch eher die Werte an MQTT Broker (bei mir ioBroker) an, trotzdem schaue ich ab und zu auch auf den ESP (yield day, Temperatur und natürlich die Uptime)

DennisXK, 2022-08-28:
Weil es dann eine runde Sache ist wenn man die Diagramme auf dem ESP sieht.

Chrischi-, 2022-08-29:
@lumapu mir ging es rein um die Visualisierung in der oben gewünschten Form. Konfiguration und live Daten wie derzeit vorhanden sind absolut sinnvoll und nachvollziehbar. 

@DennisXK 
Rund ist das in meinen Augen nicht, wenn die Daten woanders hingeschrieben und dann vom selben Device wieder aufgegriffen werden, um eine Grafik darzustellen / aufzubereiten, welche in Thingspeak sowieso vorhanden ist. Dann kann man auch direkt Thingspeak etc. aufrufen. Zudem muss in solch einem Setup das Device immer „online“ sein. Rund ist „meiner Meinung nach“, wenn du auf deinem Device selbst und zudem noch „offline“ deine Diagramme und Visualisierungen hast und das Device auch ohne Internet nutzen kannst. Aber dafür ist der ESP mit seinen limitierten Schreibzyklen nicht das ideale Device. Machbar, aber ein Tod auf Zeit ist damit vorprogrammiert …

gent99, 2022-09-03:
Ich denke nicht, dass man in ein 8266 einen Webserver einbauen kann, der in der Lage ist, die Website von ThingSpeak auszuliefern. Man könnte höchstens einen Link auf der Oberfläche platzieren, der dann zu einem entsprechenden Dashboard führt (Grafana/TS….). Den Link könnte man ja in den Settings setzen und auf der Visualisierungsseite einblenden. Aber notwendig ist das für ein Gateway wie Ahoy meiner Meinung nach nicht.

stefan123t, 2022-11-07:
@lumapu analog zum Dateneingang / -abfrage für Gen2 / Gen3 von den MI- / HM-Wechselrichtern geht es hier m.E. um eine Abstraktion der Datenausgabe via MQTT, JSON, Serial Console, ThingSpeak, SD-Card, OLED/Nokia Display etc. pp.
Wir sollten also eine generelle Möglichkeit vorsehen über verschiedene Ausgabe-Plugins die Endpoints anzusprechen die mit verschiedenen Formaten die aktuellen Daten der Inverter geschickt bekommen.

Typische Ausgabe-Formate und Plugins:
* MQTT: HomeAssistant, IOBroker, etc.
* JSON: InfluxDB, ThingSpeak, Nextcloud SensorLogger, etc.
* Text: Serial Console, OLED, Nokia Display
* CSV: SD-Card

stefan123t, 2023-01-18:
Back to the initial request:

Thingspeak from MathWorks/Wolfram Research offers two options to write Data into a so called channel:
* [REST API](https://www.mathworks.com/help/thingspeak/rest-api.html)
Use REST API calls to create and update ThingSpeak channels and charts
* [MQTT API](https://www.mathworks.com/help/thingspeak/mqtt-api.html)
Use the MQTT API to update ThingSpeak channels

The two most likely commands we would need to either implement are documented here: 

* Write Data
Update channel data with HTTP GET or POST
https://www.mathworks.com/help/thingspeak/writedata.html

```c
POST api.thingspeak.com/update.json
Content-Type: application/json
{
    "channel_id": 266256,
    "api_key": "XXXXXXXXXXXXXXXX",
    "created_at": "2018-09-10T17:41:59Z",
    "entry_id": 2,
    "field1": "123",
    "field2": null,
    "field3": null,
    "field4": null,
    "field5": null,
    "field6": null,
    "field7": null,
    "field8": null,
    "latitude": null,
    "longitude": null,
    "elevation": null,
    "status": null
}
```

* Bulk-Write JSON Data
Write many entries to channel in JSON format with single HTTP POST
https://www.mathworks.com/help/thingspeak/bulkwritejsondata.html

```c
POST /channels/999900/bulk_update.json HTTP/1.1
Host: api.thingspeak.com
Content-Type: application/json
{
	"write_api_key": "JNWIVP3CD54ZGI9X",
	"updates": [{
			"created_at": "2018-01-30 10:26:26 -0500",
			"field1": 1500,
			"field2": 150,
			"field3": 100,
			"field4": 100,
			"field5": 100,
			"field6": 100,
			"field7": 100,
			"field8": 100,
			"latitude": 123,
			"longitude": 23,
			"elevation": 34
		}
	]
}
``` 

As you can see we would need to set and store both the Channel Id and WRITE_API_KEY for this to work within our settings.
Using the Bulk Update Write Data endpoint we can give the Channel ID in the URL whereas in the single Write Data call we give it in the JSON Body.

```c
POST api.thingspeak.com/update.json
POST https://api.thingspeak.com/channels/<channel_id>/bulk_update.json
...
[    "channel_id": 266256,]
    "write_api_key": "WRITE_API_KEY",
```

Also note that the Bulk Update Write Data command will be lighter on the response side, as it only returns a success code.

```c
{
    "success": true
}
```

lumapu, 2024-08-16:
Diagram was added many versions ago, so I will close this issue now.

closed 2024-08-16 by lumapu