lumapu/ahoy #190: Feature Request: prometheus metrics endpoint

avendretter, 2022-08-21:
Hey everyone,

(german version below)
I am using Prometheus and Grafana to monitor various things at home, including my energy consumption. 
Having an awesome dashboard is one thing but having those values in a time-series database would make everything even better.

At the Tasmota project, they have a possibility to enable an endpoint, so Prometheus can scrape the data 
https://github.com/arendst/Tasmota/blob/c17e73aa9b7d2fe36d2260f59cf9a34a4263ce1d/tasmota/tasmota_xsns_sensor/xsns_75_prometheus.ino

I do not know C++ very well so I am not very confident, that I would be able to include it to this repo.

---------------------------------------------------------------------------

Hallo ihr lieben,

Ich verwende Prometheus und Grafana, um verschiedene Dinge zu Hause zu überwachen, einschließlich meines Energieverbrauchs.
Ein tolles Dashboard zu haben ist eine Sache, aber diese Werte in einer Datenbank zu haben, würde alles noch besser machen.
#GrafanaDashboards ;) 

Beim Tasmota-Projekt besteht die Möglichkeit einen Endpunkt zu aktivieren, damit sich Prometheus die Daten abholen kann
https://github.com/arendst/Tasmota/blob/c17e73aa9b7d2fe36d2260f59cf9a34a4263ce1d/tasmota/tasmota_xsns_sensor/xsns_75_prometheus.ino

Ich kenne mich mit C++ nicht sehr gut aus, daher bin ich nicht sehr zuversichtlich, dass ich das feature alleine in dieses Repo aufnehmen könnte.

aschiffler, 2022-09-08:
Maybe the /json endpoint help or can be adapted to delivery prometheus conform schema.

closed 2022-09-08 by aschiffler

avendretter, 2022-10-17:
work in progress :) 
https://github.com/lumapu/ahoy/pull/356