lumapu/ahoy #193: increase numb of inverters greater then 4 => setup error

DanielR92, 2022-08-22:
When you change the line 46 in the config.h file, from 3 to 6. The setup page is later broken. See image.
[image]

When its 4, then it work. 

Found by 'Skusi': Discord chat session.

DanielR92, 2022-08-22:
Testes with 0.5.15 (dev)

tastendruecker123, 2022-08-22:
Dem ESP8266 geht an dieser Stelle aufgrund der großen Strings der Arbeitsspeicher aus. Der ESP32 kann die Seite auch mit 6 WR darstellen. Hier das Debug Log vom 8266:

20:21:36.560 > [String] Reallocating large String(9290 -> 9308 bytes) '<p class=" ... 1" value="'
20:21:36.569 > [String] Reallocating large String(9308 -> 9370 bytes) '<p class=" ... ngth="16">'
20:21:36.575 > [String] Reallocating large String(9370 -> 9388 bytes) '<p class=" ... 2" value="'
20:21:36.583 > [String] Reallocating large String(9388 -> 9450 bytes) '<p class=" ... ngth="16">'
20:21:36.588 > [String] Reallocating large String(9450 -> 9468 bytes) '<p class=" ... 3" value="'
20:21:36.597 > [String] Reallocating large String(9468 -> 9474 bytes) '<p class=" ... ngth="16">'
20:21:36.605 > :oom(9488)@WString.cpp:195

Das "oom" dürfte wohl für "out of memory" stehen.

Skusi, 2022-08-22:
Keine Chance das noch in den 8266 zu bekommen ? Kann man irgendwo noch was einsparen ?

lumapu, 2022-08-22:
bin dran den AsyncWebserver einzubauen, ich denke der kann das dann wieder. Bitte noch etwas Geduld.

tastendruecker123, 2022-08-22:
Hat eher damit zu tun, wie der HTML Code intern produziert wird. Das ist momentan sehr speicherintensiv da viele String Objekte erzeugt werden. Wenn man das umschreibt, wird es auch mit mehr Invertern gehen. Oder anders ausgedrückt, es liegt gar nicht mal so sehr an der Inverter-Anzahl sondern eher an der Größe des HTML Codes von /setup und dessen Erzeugung.

lumapu, 2022-08-22:
genau, deswegen gibt es in Zukunft das hier:

https://github.com/lumapu/ahoy/blob/asyncWeb/tools/esp8266/tmplProc.h

aschiffler, 2022-08-22:
@lumapu Vergiss nicht yield() in den callbacks bzw. den bebundenen "server.on" Funktionen raus zunehmen. Sonst läuft der webservice nicht. richtig "stabil". Das ist meine Erkenntnis. Z.B. in /livedata.
Ich habe heute auch den Async eingbaut. Habe dann alles gestasht und wieder zurück auf los.
Meine Erkenntnis, der AsyncWeb macht nur Sinn wenn wir die html Seiten in ein File-System legen. Sonst wird das unpflegbar.
Ich habe nicht wie du einen "tmplProcessor" genutzt sondern direkt alles einfach umgeschrieben.
Meistens mit request->send_P(200,html.c_str()) --> geht laut Doku bis 1400 Byte für "html". Also ohne streamresponse.
Aber diese Funktion ermutigt weiter dran zu bleiben:
https://github.com/me-no-dev/ESPAsyncWebServer#setup-event-source-on-the-server

lumapu, 2022-08-22:
@aschiffler Danke für die Hinweise. Ich glaube dass yield wird von AsyncWebserver auch als Warnung gemeldet.
Wie läuft das mit HTML im Filesystem? Statische Seite mit Javascript, die verzögert über die Events die Daten nachläd? Das ist auf jeden Fall ein ganz neuer Ansatz und definitiv interessant.

aschiffler, 2022-08-22:
@lumapu Ja genau. Statische Seite html +Java Script und dann via Xhr z.b. einen SSE Endpunkt der im Javascript mit einem Eventlistener überwacht und direkt verwendet wird.
Siehe OpennDTU

stefan123t, 2022-08-24:
Would be good to add a #pragma error in case we have more than 4 and ESP8266

@aschiffler is there a similar maximum for ESP32 at the moment ?

tastendruecker123, 2022-08-24:
I'm currently working on a more streamlined version of /setup that doesn't consume much RAM. It's still rough around the edges and not as easy to deploy right now (because the LittleFS filesystem needs to be built and uploaded separately), but I have put it up on my repo (in /dev2). At least on the web interface side there's no practical limit on the number of inverters, and the setup page works fine with 12 inverters on an ESP8266.

stefan123t, 2022-08-24:
@tastendruecker123 just read up on the discord what you are about to do. So the LittleFS can be uploaded / updated via Web Interface too or is this only possible via the tools in pio ?

I recall that some allow to upload both firmware and filesystem images but I dunno how it is done as the firmware.bin is stored in the (SPIFFS?) filesystem until flashed.
There is mention of AsyncElegantOTA to be capable of also uploading some files to the (SPIFFS) filesystem.

https://randomnerdtutorials.com/esp32-ota-over-the-air-arduino/

Though I do not know whether AsyncElegantOTA would equally well work with LittleFS nowadays and tutorials are just outdated or whether we need to stick with SPIFFS ?

Note: Sticking with SPIFFS may not be a bad decision in the first place if we are trying to append data to files as it is not using copy-on-write as LittleFS which would prevent lots of 4k block writes in such a use case.

See #164 for filesystem decision and #185 for the logging / append to file use-case. Thanks

DanielR92, 2022-09-22:
Are we at now on a point, to close this issue? 
From my site it was only a hint for a "bug"... 

Also it is possible to write a secure part in, to limit in a ESP8266 to until 4 Inverter. 
When it is a ESP32, then we can increase the number.

What do you think about it?

lumapu, 2022-09-22:
we should test it again using the development branch with AsyncWebserver.
For now the maximum number is set to 4 but not testet with grater numbers.

DanielR92, 2022-09-23:
Next week I can test it (not with real inverter). Thanks for reply. 👍

stefan123t, 2022-09-23:
I think this calls for SPIFFS/LittleFS storage just as #164 for storing the settings. This has some details about how to bundle / upload the bin and the filesystem image.

dennsen79, 2022-10-12:
Hallo,

ich brauche ein fertige Firmware für meinen 8266 mit 6 Invertern. In der Version 0.5.17 gehen ja derzeit leider nicht mehr. Kann mir jemand eine Firmware bauen mit 6 Invertern in einer Version mit der es laufen würde?

Danke im Vorau!

DanielR92, 2022-10-12:
Moin,

habe es gerade getestet. Also es sollte nun klappen:
[image]

Da ich nicht so viele WR besitze, müsste das jemand anders testen.

dennsen79, 2022-10-14:
Hallo,

Ich konnte die Firmware 0.5.19 mit fünf Invertern (4 x HM 300, 1 x HM 600) testen. Leider habe ich Probleme mit den empfangenen Daten über MQTT. Werte werden falsch dargestellt.

Hat jemand auch schon diese Version mit mehr als vier Geräten ausprobiert?

Kann mir jemand eine funktionierende Firmware für bis zu sechs Geräten oder genau fünf zur Verfügung stellen? Am besten die Version 0.4.25

Alternativ würden mir auch die Sourcen zur Version 0.4.25 reichen.

Vielen Dank im Voraus!

lumapu, 2022-10-31:
bezüglich MQTT bitte nochmal mit aktuellem Release 0.5.28 testen
Die Firmware unterstützt 4 Inverter

stefan123t, 2022-11-07:
@dennsen79 und @DanielR92 könnt Ihr mal das Issue mit dem aktuellen Release 0.5.28 oder Development 0.5.32 prüfen, wieviele Inverter da als Maximum gehen ?

@Skusi und @tastendruecker123 / baloo wir haben den AsyncWebServer ins o.g. Release mit eingebaut. Willst Du das Issue eventuell nochmal bewerten ?

lumapu, 2022-11-15:
in der aktuellen development Version gehen bis zu 10 Inverter

DanielR92, 2022-11-16:
Kann das jemand testen?

Skusi, 2022-11-16:
Ich hab die 0.5.37 seit gestern mit 6 Invertern erfolgreich laufen !

dennsen79, 2022-11-17:
Ich habe die 0.5.37 seit gestern mit 5 Invertern (4 x HM 300,  1 x HM 600) erfolgreich am laufen!

stefan123t, 2022-11-17:
@lumapu ich glaube Du kannst das ins nächste Release übernehmen und hier zu machen.

closed 2022-12-01 by lumapu