lumapu/ahoy #196: V 0.5.15 stable? Einige Fehler?

CBRocker, 2022-08-23:
Hallo,
ich habe gestern die ahoy 0.5.4 auf meinen ESP geflasht und entsprechend meine Daten eingetragen. Hat alles
wunderbar funktioniert und alle 15 sek aktualisiert.
Dann habe ich gesehen, daß es die 0.5.15 gibt. Diese habe ich dann erst per wlan über den Punkt "Update Firmware" in der WebUI aktualisiert. Da ich (verständlicherweise) dann keine Verbindung mehr hatte, habe ich mich mit dem Handy verbunden und alles neu eingerichtet. Aber die WebUI hat dann sehr komisch ausgesehen, siehe Screenshots. Außerdem leuchtet seit dem Update die blaue LED des ESP nicht mehr.
Ich habe dann nochmal per ESP Flasher geflasht, aber hat nichts gebracht, die 0.5.15 ist nach wie vor (bei mir) instabil und funktioniert nicht.

Ich habe dann wieder die 0.5.4 geflasht. Nun läuft wieder alles. 
Dazu aber diese Frage:
Bei General - Interval habe ich 20 sek eingetragen, bei MQTT - Interval habe ich 80 sek drin.
Die WebUI wird nun alle 80 sek aktualisiert. Welcher Intervall ist denn nun der primäre?

Weitere Frage:
Bei den Statistics habe ich 576 Receive Success und 1681 Receive fail (Zeitraum 12 Stunden).
Ist die Anzahl Receive fail ok/normal/unnormal?

Nächste Frage:
Bei Alarm_Mes_ID habe ich 10. Was bedeutet das?
(Verbesserungsvorschlag: Falls die Zahl nie Dezimalstellen haben wird, könnte man diese ja auch weglassen, sieht vielleicht besser aus)

Tolles Projekt und mit der Anleitung war es supereinfach, alles zum Laufen zu bringen.
Danke an den/die Entwickler.

Grüße Christoph
[image]
[image]
[image]
[image]

aschiffler, 2022-08-24:
Hi,

you have to erase the flash before updating. you can do this partly by click "erase settings" in the setup.
AlarmMesId is a counter. It shows the number of warnings/alarms are send by the inverter after the last off (no DC power) on (DC power available) cycle. And yes without decimal point it would by nicer.
About the intervals. Ignore the mqtt interval it will be removed in further releases. The mqtt interval is the inverter interval (in case the request is successful)
About the fails. Set the send/receive power of the radio part as low as possible. It is a hardware/radio topic.

maurice1, 2022-08-24:
The current page may be perfect for experts, for "regular" people however, the values on the visualization page could be cleaned up a little:

- Display values as integers, when they're coming as integers from the inverter (YieldDay, VAr, ALARM_MES_ID)
- Clarify labels like ALARM_MES_ID. Could be something like "errors today" or anything else more descriptive.
- internal value names like ALARM_MES_ID could still be included as the label's tooltip (html element title).

CBRocker, 2022-08-24:
Hi @aschiffler @maurice1 
thanks for these informations. I am happy to get it. I will try to flash the 0.5.15 as specified by aschiffler with erase settings.
I will set the send/receiver power lower and test it.
Thank you very much

stefan123t, 2022-08-24:
@aschiffler is there anything we could do about the bin files packaging. I imagine that we could possibly overwrite the EEPROM settings automagically before/after a firmware update. This may obly be necessary if the firmware involves a change of the settings stored after the ssid/pw. So a kind of settings version should be introduced and when we see an "upgrade" is necessary we erase the EEPROM (after the ssid/pw) accordingly.

aschiffler, 2022-08-24:
> @aschiffler is there anything we could do about the bin files packaging. I imagine that we could possibly overwrite the EEPROM settings automagically before/after a firmware update. This may obly be [...]

The Solution would be to implement an own update route for the OTA.
There we can check the compatibility and erase the Flash At the emulated eeprom Position After the wlan Settings. (Similar to the OTA route in OpenDTU)

closed 2022-09-04 by aschiffler