lumapu/ahoy #197: Configuration for NTP port

maurice1, 2022-08-24:
While it's possible to configure the NTP port on ` /setup`, the port is not used in *ahoywifi::sendNTPpacket*. I wonder if anyone would set a port other than 123 for NTP. In my opinion, it's okay to hardcode 123 and remove the port configuration on the setup page altogether.

DanielR92, 2022-08-24:
in the cpp file from ahiywifi, it exist at line 186 "mUdp->begin(mConfig->ntpPort);".
What do you mean the port is not used???

maurice1, 2022-08-24:
@DanielR92 The configuration value is used there, yes, but not in method *ahoywifi::sendNTPpacket*, where port 123 is used.

lumapu, 2022-08-25:
It's not that easy to fix. I tried to add the port from configuration to the mentioned line but my ESP was unresponsible. My current configuration has set 8888 as port. I think the ESP stucks in an endless boot loop, because the _wrong_ port is unresloveable and than the WDT fires.

maurice1, 2022-08-25:
Wouldn't it be sufficient to replace usages of the configured value with "123"? [This line](https://github.com/grindylow/ahoy/blob/348ad4bd07ddde67cd017547b2afcafa486f2b09/tools/esp8266/ahoywifi.cpp#L186) would then read:

```cpp
mUdp->begin(123);
```

Maybe I'm missing something and this is the source port. In that case, the description on the setup page would be misleading.

DanielR92, 2022-08-25:
Im sorry you are right. :)

closed 2022-11-06 by lumapu