lumapu/ahoy #198: MQTT command check data (<INVERTER_ID>)

DanielR92, 2022-08-24:
The most User setup the DTU with a Serial from the inverter. When they try to send a command in mqtt for example with the command: "<CHOOSEN_TOPIC_FROM_SETUP>/devcontrol/<INVERTER_ID>/11", they could send

A.) inverter/devcontrol/0/11 
B.) inverter/devcontrol/11611151.../11 

It is recommended to change the code "hmSystem.h" in line 90.

``` INVERTERTYPE *getInverterByPos(uint8_t pos, bool check = true) {
            DPRINTLN(DBG_VERBOSE, F("hmSystem.h:getInverterByPos"));
            if((mInverter[pos].initialized && mInverter[pos].serial.u64 != 0ULL) || false == check)
                return &mInverter[pos];
            else
                return NULL;
        }```

aschiffler, 2022-08-24:
The check shall be done in the mqtt callback.

aschiffler, 2022-09-04:
This is now fixed. The mqtt topic only supports the id of the inverter not the serial number in the topic path.

closed 2022-09-04 by aschiffler