lumapu/ahoy #201: Feature Request : Limit-Anzeige auf Website in gewählter Einheit "Prozent" oder "Watt" anzeigen

ntfrnd, 2022-08-25:
Begrenze ich meinen HM-600 auf 320 W sehe ich in der seriellen Debugausgabe auch diesen Wert, nach kurzer Zeit regelt sich der Inverter auf diesen Wert ein. Das finde ich als prima Feature.
Allerdings scheint ahoy dies intern immer in Prozent umzurechnen, denn auf der Website steht dann z.B. 53%. Dies finde ich etwas verwirrend:

[image]

Ich schlage vor, auf der Website den eingestellten Wert anzuzeigen, und zwar in Prozent oder Watt - je nachdem wie dieser eingestellt wurde. Noch hilfreicher finde ich die Zusatzangabe "persistent" oder "non persistent", so dass man mit einem Blick sieht, was konkret eingestellt wurde.

DanielR92, 2022-08-25:
Die Idee ist gut, jedoch fraglich. 
Denn Absoluten Watt hast du selbst im Screenshot, in diesem Beispiel 324.30 W (P_AC = Power AC).
Der Prozentwert ist eher reingeflossen da man es aus dem Datenstrom auslesen konnte.

Desweiteren haben wir mW nach nur die Infos was für ein Wert aktuell eingestellt wurde. Ob es sich um ein Pers. oder Temporären Wert handelt konnten wir bisher nicht rauslesen...

Gruß Daniel

ntfrnd, 2022-08-25:
Das stimmt nur teilweise. Die 324,30 W ist die aktuell erzeugte Leistung - dies kann aufgrund von Beschattung weniger sein, als was als Limit eingestellt wurde. Von daher wenig aussagekräftig.
Nachdem ich doch das Limit über mqtt sowohl in Prozent als auch in Watt einstellen kann, warum sollte man das dann nicht anzeigen können? Versteh ich nicht.

MubiTec, 2022-08-25:
Find ich auch gut.  Ist sozusagen das I-Tüpfelchen. Und in einer vorherigen Version war es auch so, ich meine in 0.5.13 oder 0.5.14.

stefan123t, 2022-08-30:
Das hört sich nach dem goldenen Türgriff an. Wir können aktuell nur den Wert in Prozent auslesen. Es gibt zwar Ideen den Inverter anfangs kurzzeitig auf 100W zu stellen um die tatsächliche max Leistung des Inverters zu ermitteln aber ohne diese Angabe können wir die %-Werte nicht in eine plausible Watt Angabe zurückwandeln. Ergo kann das Feature nicht ohne größeren Aufwand implementiert werden. Vor allem können wir nur dann eine Watt Angabe zurückgeben wenn wir wissen ob vorher Prozent/Watt gesetzt wurde und der ESP oder der WR (wegen Tageslicht) sich seitdem nicht resettet hat.
Wie bereits beschrieben ist auch die Information temporär/permanent aktuell nicht auslesbar.
Ich schlage daher vor das Issue vorerst zu schliessen bzw wenn dann nur die aktuelle gesetzte max Leistung/Limitierung in Watt über einen anderen Endpoint zurückzugeben.

stefan123t, 2023-01-12:
@lumapu OpenDTU hat das als Feature eingebaut. 
Thomas rechnet dafür mit dem Max Power der einzelnen Modelle, die rekombiniert er aus der HW_VER Versionsnummer.
Vielleicht wollen wir das Issue wieder aufmachen und es irgendwann analog zur OpenDTU umsetzen ?

closed 2023-12-29 by lumapu