lumapu/ahoy #206: Irradiation at 0%

gent99, 2022-08-26:
Irradiation value at my panels is always 0%. Is this a bug or is it not implemented/reverse engineered yet?
Kind regards

duenengeist, 2022-08-26:
Maybe a stupid question, but did you enter the Max Module Power (Wp) in the setup? This is used to calculate the irradiation.

I run AHOY - 0.5.14 and it does show irradiation-levels which make sense to me

lumapu, 2022-08-26:
correct, for the latest version (release) 0.5.15 the irradiation works as expected.
The only improvement from code side could be a check whether the irradiation is not equal to zero before displaying the value.

gent99, 2022-08-27:
That was not known to me. I did not yet set the Max Module Power in Setup. I think there's no code change necessary.

Thank for answering. Kind regards

closed 2022-08-27 by gent99