lumapu/ahoy #207: HM-300

siegfriedhoellrigl1, 2022-08-28:
Should the ESP8266 Software work with the HM-300 Inverter too ?

I have tried this with SW Version 0.4.19, 0.4.22 and with 0.4.25.
(because there are binary downloads on the internet for that...)

It seems that there es some interaction between the ESP and the RF Module.

But no values are recieved from the inverter.

Chrischi-, 2022-08-29:
Look under releases for the 5.15

https://github.com/grindylow/ahoy/releases

Matz3e, 2022-08-29:
Yes it's working with the HM-300. My ahoy 5.15 is running perfect. Try to go very close the the HM for the fist time.

DirkausHH, 2022-08-29:
0.5.13 is working stable for some days with my HM-300.
In the beginning it took up to 30 minutes to receive first data, but now it's running fine.

siegfriedhoellrigl1, 2022-08-29:
Would someone be so kind, and post a binary version of 0.5.13 ?
It seems to be a bit challenging to compile it.

fila612, 2022-08-29:
please have a look at the "Actions" section on this page.

aschiffler, 2022-09-04:
Close due to inactivity. Assume issue is fixed.

closed 2022-09-04 by aschiffler