lumapu/ahoy #209: Ungenaue Uhrzeit / NTP nur beim Start

thfcm, 2022-08-29:
Problem:
Die Uhrzeit auf dem Ahoy ist ungenau und weicht schnell ab 

Ursache:
Aktuell wird die Uhrzeit via NTP nur beim Start geholt und im Anschluss wird sich auf einen Zähler verlassen.

lumapu, 2022-08-29:
@aschiffler das sollten wir noch fürs 0.5.16er Release übernehmen

lumapu, 2022-08-30:
@stefan123t schaue mal in Discord von gestern Abend.

Intervall sollte 12h oä sein

thfcm, 2022-08-30:
Folgender Vorschlag - ungetestet und bei Google gefunden:

unsigned long seconds= 0;
unsigned long prevMillis = 0;

if (millis() - prevMillis >= 1000) {
    prevMillis += 1000;
    seconds ++;
}

Hier sollte der Überlauf irrelevant sein und im schlimmsten Fall alle 50 Tage für 999ms Abweichung sorgen. Da sind die Toleranzen der Keramischen Schwinger vermutlich deutlich schlimmer.

Wenn man seconds beim Erhalt der NTP-Zeit auf 0 setzt, hat man die Differenz zur letzten korrekten NTP-Zeit...und das im schlimmsten Fall für ganz viele Jahre... z.B. wenn die Zombie-Apokalypse doch noch kommt.

Und ansonsten könnte man z.B. alle 24h die NTP-Zeit holen zwecks Abgleich.

homeautomation2022, 2022-08-30:
Ich wäre für einen NTP Abgleich täglich um 4 Uhr, also noch vor Sonnenaufgang, aber bereits nach der Zeitumstellung(DST), die dort gleich mit geprüft wird.
Alternativ könnte man auch einen Timer für den nächsten Sync nutzen, wie es viele fertige Libs machen.

gent99, 2022-09-03:
Welcher Port ist denn konfiguriert? 8888 als Vorgabe beim Setup stimmt nicht. NTP ist UDP auf Port 123. Sobald man den einträgt, ist die Zeit so wie auf dem Timeserver.

lumapu, 2022-09-04:
das heißt, wir müssen einfach den Default ändern?

commit b67cd033 (2022-08-30): fix #209
closed 2022-09-04 by aschiffler