lumapu/ahoy #214: read only?

drahdiwaberl, 2022-08-30:
so weit ichs verstehe liegt der Fokus dieses Projekts bisher nur auf dem Auslesen von "Telemetrie"Daten der Wechselrichter, oder liege ich falsch?

gibt es Ambitionen und die Möglichkeit auch schreibende Methoden zu implementieren, also zB die Einstellung der Exportleistung?

duenengeist, 2022-08-30:
Hallo,

lies mal ein wenig in diesem Thread hier, das sollte diene Fragne beantworten:

https://www.mikrocontroller.net/topic/525778

Ja, es ist möglich die Einspeiseleistung einzustellen...

LG

Carsten

drahdiwaberl, 2022-08-30:
gut, ein wenig versucht aus dem megathread etwas rauszufinden - es scheint also in der esp-version sogar im UI implementiert? (ich hab meinen nrf am raspi hängen und nutze momentan das script unter tools/rpi)

https://github.com/grindylow/ahoy/issues/31

oh. und da jetzt auch doku entdeckt.. 
https://github.com/grindylow/ahoy/blob/main/tools/esp8266/User_Manual.md#active-power-limit-via-mqtt

na vielleicht switch ich doch auch auf nen esp..

mslookup, 2022-08-30:
Begrenzung der Exportleistung funktioniert seit https://github.com/grindylow/ahoy/commit/348ad4bd07ddde67cd017547b2afcafa486f2b09 (0.5.15) einwandfrei, sowohl absolut als auch relativ in % einfach über die UI im Bereich "Setup" einstellbar.

VG

DanielR92, 2022-08-31:
Im Rpi kann man das ganze weiterentwickeln, das wollte ich Ursprunglich auch machen. Habe es aber auf Eis gelegt, da viele nur ESP nutzen.

drahdiwaberl, 2022-08-31:
ganz klar dass die meist genutzte plattform am weitesten entwickelt ist, weil halt auch die meisten neu dazu kommenden dazu am meisten infos finden. 

mir persönlich war halt die Zuverlässigkeit vom ethernet am raspi - auf dem sowieso die homeautomation auch läuft - wichtiger als ein UI. und ESPs fliegen inzwischen so viele im Haus herum dass ich inzwischen auf die zentrale Verwaltung von esphome nicht mehr verzichten will/kann..

also ja, ein kompletteres command-set via mqtt in einem tool am raspi wär natürlich cool - aus meiner sicht bräuchte es da auch nicht unbedingt UI&co weil die Nutzergruppe wohl eine andere ist..

aschiffler, 2022-08-31:
Ich schließe das issue. Diskussion gerne auf discord.

closed 2022-08-31 by aschiffler