lumapu/ahoy #225: TOR_Erzeuger definieren

mesingerold, 2022-09-01:
Hallo,

vorab ein großes Dankeschön an die Entwickler - super Arbeit 👍

System läuft unter v 0.5.15 perfekt, optimal wäre noch die Möglichtkeit einer Definition des Standard TOR_Erzeuger-Profils (Bsp: AT_TOR_Erzeuger_default) via Interface!!

Vielleicht findet ihr eine Möglichkeit,

Danke & LG

stefan123t, 2022-09-01:
Siehe #96

DanielR92, 2022-09-22:
Das müsste man hier diskutieren. Ich bin alles nur überflogen... habe ich richtig gelesen, das es die Möglichkeit gibt in der orig. DTU via Auswahl ein Profil zu aktivieren, der WR sich dann daran anpasst?

Oder geht es darum komplette Config des WR zu modifizieren?

stefan123t, 2022-09-23:
@DanielR92 in der DTU Pro kannst Du ein sog TOR / GridProfile auswählen. Dieses wird auf den WR geladen und dann arbeitet der WR nach den Anweisungen des GridProfiles. Dieses enthält ua die nationale Netzfrequenz, das obere und untere Netzspannungslilit etc pp.
Das müsste also De-ichirou einmal umstellen und tracen damit wir die Kommandos und Antworten vom WR haben.
Das ganze gehört aber mE bereits in das andere issue und braucht hier nicht doppelt diskutiert werden.

stefan123t, 2022-10-13:
Bitte in #96 weiter diskutieren Danke

stefan123t, 2022-10-14:
closed as duplicate of #96

closed 2022-10-14 by stefan123t