lumapu/ahoy #226: HM-400 fehlende Werte YieldDay, YieldTotal

DennisXK, 2022-09-01:
Ich habe heute einen weiteren Wechselrichter HM-400 hinzugefügt. 

Mir ist aufgefallen das AC-seitig keine YieldTotal und YieldDay angezeigt wird.

kann das bitte gefixt werden!? Denn ohne die Angebe ist die Abfrage sinnlos😢

stefan123t, 2022-09-02:
@DennisXK kannst Du uns ggf ein TX/RX log mit den DEBUG in dbg.h aktiviert oder einen Dev Debug Build schicken ?

stefan123t, 2022-09-02:
Interessant is auch dass das Limit auf 65535 = 0xFFFF gesetzt ist @aschiffler ist das eventuell die Ursache.
@DennisXK Kannst Du mal 400W / 100% als Power Limit setzen ?

aschiffler, 2022-09-02:
Hi,
kannst Du bitte ausschließen ob es daran liegt dass der zweite WR gar nicht antwortet.
--> Lösche den  ersten und füge ausschließlich den zweiten hinzu.

DennisXK, 2022-09-02:
Ja, werde ich heute versuchen, aber mit der 0.4.x version habe ich generell Werte erhalten, bloß halt nicht YieldDay und YieldTotal.

Limit habe ich keins einsgestellt. sobald der Wechselrichter produziert geht er auf 100%. so war es bei meinem HM-600. Hatte den HM-400, erst gegen Abends eingerichtet, deswegen sind alles Werte auf null.

DennisXK, 2022-09-02:
kann leider kein Bilder mit meinem Iphone posten 🤷‍♂️

[image]

stefan123t, 2022-09-02:
@DennisXK 0.4.x hatte evtl noch Probleme mit einzelnen WR Modellen. Probiere doch mal 0.5.16 oder gleich die Developer / Debug Builds ?

DennisXK, 2022-09-02:
ich habe die 0.5.16 drauf, bei dieser Version zeigt er mir keine YieldDay und YieldTotal auf der Ac Seite an. Mit dem Bild von meinem letzten Post habe ich gezeigt das er Werte erhaltet mit der 0.5.16 Version.

Wenn du mir genau sagst was ich machen soll, kann ich das machen um das Problem einzugrenzen.

DennisXK, 2022-09-03:
@stefan123t was soll ich jetzt genau machen , das wir das Problem beheben können?

die debug Version flashen und dann? Was brauchst du von mir?

aschiffler, 2022-09-04:
Hi,
kannst Du bitte ausschließen ob es daran liegt dass der zweite WR gar nicht antwortet.
--> Lösche den ersten und füge **ausschließlich** den zweiten hinzu.
--> Wie ist dann die Fkt.?

DennisXK, 2022-09-04:
Hi,
@ashiffler das habe ich in dem Post vom 2 Sep. 11.30 gezeigt. Beide Wechselrichter liefern Werte. Aber ich lösche dwn HM-600 zum test raus!

DennisXK, 2022-09-04:
@aschiffler 

nur der HM-400
[image]
HM-600 und HM-400
[image]

Wie oben schon geschrieben hatte der HM-400 keine Werte angezeigt, da ich ihn Abends (dunkel) hinzugefügt hatte. Des Weiteren zeigte er auch bei der Limitanzeige 65535% an, dass ebenfalls verschwand am nächsten Tag als es hell wurde. Das sollte man evtl. auf fixen. Denn 65535% kam mir sehr merkwürdig vor.

Das Problem besteht weiterhin das YieldDay und YieldTotal nicht angezeigt wird, egal ob nur der HM-400 oder beide hinzugefügt wurden.

aschiffler, 2022-09-04:
Ok jetzt ist es deutlich.
Du hast nur ein Panel am zweiten hm400.
Schreibe einfach ein zweites dumme Panel ins Setup rein.

DennisXK, 2022-09-04:
Der HM-400 hat nur einen Anschluss also nur einen MPP-Tracker. Der HM-600 hat 2 MPP-Tracker

aschiffler, 2022-09-04:
Ah. Ok da war ich etwas auf dem Schlauch gestanden.
Ich habe im development branch nun für den Parser die Felder hinzugefügt. Allerdings ungetestet. D.h. teste es mal bitte mit der Firmware hier:
https://github.com/grindylow/ahoy/suites/8126656330/artifacts/351967627

DennisXK, 2022-09-04:
@aschiffler keine verbesserung YIeldDay und YieldTotal werden weiterhin nicht auf der AC Seite angezeigt.

aschiffler, 2022-09-04:
AC Seite ist für die das grüne Feld richtig?

aschiffler, 2022-09-05:
@DennisXK 
Ich kann das bei mir nicht nachstellen. Bei mir wird das Feld oben ("grüner Bereich") ebenfalls angezeigt. (Development Build)
Es gab noch ein Fehler im Parser für den HM400. Ggf. behebt dies das Problem. 
Siehe: https://github.com/grindylow/ahoy/commit/03a6b87cc1d889d6082c0b4e088d530f8990f914

Bitte nochmals testen --> Actions --> Firmware aus dem letzten Run

Wenn es immer noch nicht geht bitte Issue wieder öffnen.

closed 2022-09-05 by aschiffler