lumapu/ahoy #228: HM-600 auf 800w bringen?

setje, 2022-09-02:
Hallo zusammen,

ich habe verstanden, dass sich das Limit des HM-600 auf 800w statt 600w bringen lässt. Es scheint, als wäre der String bei 320w limitiert zu sein. Zumindest sehe ich das bei dem String, wo ich 2x 335w Module per Y-Adapter dran hängen habe, die jeweils in Ost/West ausgerichtet sind.  Wenn ich das Power-Limit auf z.b. 125% erhöhe, bleibt der String bei max. 320,40w stehen.
Mache ich etwas falsch bzgl. Power-Limit? Folgendes habe ich eingestellt:
[image]

duenengeist, 2022-09-02:
Hallo,
hat er die 125% denn akzeptiert? Afik werted die neueste Version doch die Rückmeldung des WR aus, oder?

setje, 2022-09-02:
Hallo, wie kann ich das Prüfen? Im Visualization Frontend steht In der Überschrift: "HM-600 Limit 125 %"
[image]

duenengeist, 2022-09-02:
OK, dann glaub ich mal auf der seriellen Console mitlesen. Habs bei mir auch probeweise grad mal eingestellt. Habe aber leider keine serielle dran, da auch "Funkgründen" der ESP im Gartenhaus montiert ist.

lumapu, 2022-09-02:
evtl. bist du am Limit des Channels? Ich glaube nicht, das ein HM600 seine gesamte Leistung über einen Eingang beziehen kann.
Mit welchem maximalen Strom ist der HM600 abgegeben?

duenengeist, 2022-09-02:
Stimmt angegeben

[image]

sind 11,5A und wenn das Panel im MPPT vielleicht 36V hat sind das 400W per Channel. War hier nicht jemand, der eine 48V-Batterie dran hat, der könnte das ggf. einfacher testen. Wobei ich auch die Gefahr sehe, dass etwas kaputt geht...

lumapu, 2022-09-02:
und wenn die Eingänge zu unterschiedlich sind? Der hat doch nur einen MPPT oder?

duenengeist, 2022-09-02:
der hat m.E 2 MPPT, das ist an den unterschiedlichen Spannungen zu sehen (grade sind bei mir beide voll im Licht (Wolken), daher nicht viel Differenz). Biem HM-1200 sind je zwei zusamengeschaltet, da sind die Spannunge dann immer genau gleich.

[image]

lumapu, 2022-09-02:
Den HM1200 habe ich auch. Dass die Spannungen hier identisch sind liegt an meiner Implementierung. Wir haben in der Payload nur 2 Spannungen gefunden, daher würde jede kopiert um für alle Channels eine Spannung zu haben.
Ich habe allerdings auch getestet, dass zB. Eingang 1 und 2 auf die erste Spannung wirken.

duenengeist, 2022-09-02:
Ich könnte das die nächsten Tag noch mal am Gerät nachmessen...

stefan123t, 2022-09-02:
@duenengeist bisher war kein Limit über 100% möglich es sei denn es hat sich in den letzten Wochen hierzu Neues ergeben. Vielleicht mal mit @klahus1 oder @aschiffler  im Chat besprechen die hatten das ActivePower Limit ausgetestet und implementiert.
@aschiffler hatte mW auch eine Option eingebaut die aktuellen Settings in Prozent aus dem 0x05 SystemConfigPara auslesen.

aschiffler, 2022-09-02:
Ja, das aktuell gesetzte Limit wird jedes mal wenn man es ändert ausgelesen. Also quasi geprüft ob es "angekommen" ist.

Chrischi-, 2022-09-04:
@stefan123t 
im DEV-Branch sind derzeit auch keine 125 % möglich. Gerade mit dem Build von gestern getestet.

Mich würde aber auch interessieren, wie man dem Mikro-Inverter (HM-600) sagen kann, welche Panels
dran hängen und er somit nicht direkt bei jeweils ~320 W reguliert.

duenengeist, 2022-09-04:
das Web-UI zeigt aber 125% an, siehe mein Screensaht oben...

Chrischi-, 2022-09-04:
@duenengeist 
Ja alles gut! Mir ging es um die Entwicklungsversion, bei der es in meinem Fall nicht geht. 

Du nutzt ja die 5.15, oder?

duenengeist, 2022-09-04:
ja, ich nutze die 5.15

closed 2022-09-04 by aschiffler