lumapu/ahoy #230: NRF Module

billy0xff, 2022-09-03:
[documentation]

folgende NRF24L01+ Module kann man via Aliexpress um kleines Geld beziehen, eines funktioniert im RX nicht, das andere funktioniert einwandfrei, gestestet mit der aktuellen release.
[image]

helgeschneider, 2022-09-03:
Sind das die fuer 10 Stueck 10€ incl. Versand? Die hab ich gerade bestellt, verdammt. 
Habe aber das gleiche Design von Ebay (schwarz) und die tun einwandfrei.

billy0xff, 2022-09-03:
Den Unterschied macht das Marking am Chip: der gute hat links über der Beschriftung ein gefülltes Quadrat, der schlechte einen gefüllten Kreis.  Das Pinout der Module ist auch unterschiedlich!
Das gute habe ich von dort: [https://de.aliexpress.com/item/1005004221551392.html](url)
Pinouts:

`NRF24L01+
         _   ___
        | | | 
        | |_|     
        |       
        |-----o       

           +---+
           |   |
           |o  |
           +---+

       1 OOOOOOOO 8
         ||||||||
         |||||||+-- 8 IRQ
         ||||||+--- 7 MISO
         |||||+---- 6 MOSI
         ||||+----- 5 SCK
         |||+------ 4 CSN
         ||+------- 3 CE
         |+-------- 2 GND
         +--------- 1 Vcc

-> NOK (Fake-Chip?)

NRF24L01P  (Pinout!)
https://de.aliexpress.com/item/1005004221551392.html
         __    ___
        |  |  | 
        |  |__|     
        |       
        |
           +---+
           |   |
           |=  |
           +---+

       1 OOOOOOOO 8
         ||||||||
         |||||||+-- 8 GND
         ||||||+--- 7 IRQ
         |||||+---- 6 MISO
         ||||+----- 5 MOSI
         |||+------ 4 SCK
         ||+------- 3 CSN
         |+-------- 2 CE
         +--------- 1 Vcc

-> OK!`

closed 2022-09-04 by aschiffler

lumapu, 2022-09-04:
ist die Info in ein readme geflossen?

billy0xff, 2022-09-05:
HM400 (FW-Version: 10014)

Grufbert, 2022-09-06:
Here's the difference: The NRF24L01+ is a newer version of the NRF24L01, capable of doing an extra 250kbps of on-air data rate while the one without “+” has only 1Mbps and 2Mbps. Both versions can be mixed together as long as 1 or 2 MBps is being used as the data rate.

Soweit ich das Projekt verstehe benötigt die Kommunikation genau die 250kbps.

lumapu, 2022-09-06:
richtig, der Wechselrichter schreibt die Bitrate vor

stefan123t, 2022-09-07:
@lumapu ich kann mich erinnern dass die DTU Pro zwischen 1/2MBps und 250kBps wechselt. Es gibt sogar ein extra NET_ Kommando dafür. Eventuell ist das ja die Ursache dass manche keinen Empfang bekommen ?

flummy1978, 2022-10-23:
Hallo zusammen,
ich möchte mir das Teil auch zusammenbauen und bin auf dieses "Problem" gestoßen ... 

> Ich verwende einen NRF mit "kreis" und dieser funktioniert.
@Grufbert  Musstest Du dafür irgendwas anders machen, oder war lediglich das Kompilieren das Problem? Bzw welche Pinbelegung hast Du genommen? 

Bzw hat jemand eine Bezugsquelle in Deutschland (und Umgebung) wo man den richtigen Chip bekommt? 

Vielen Dank im Voraus
VG
Andreas

cbscpe2, 2022-11-08:
Ich habe die NRF24L01 Module WPI322 von velleman whadda und musste gerade feststellen, diese haben auch den Chip mit dem ausgefüllten Kreis und wenig überraschend empfange ich nichts von meinem Hoymiles arrrgghhh. Ich wäre auch über eine Bezugsquelle die zuverlässig den richtigen Chip verbaut dankbar.

Gruss

Peter

Grufbert, 2022-11-08:
> Hallo zusammen, ich möchte mir das Teil auch zusammenbauen und bin auf dieses "Problem" gestoßen ...
>
>
> @Grufbert Musstest Du dafür irgendwas anders machen, oder war lediglich das Kompilieren das [...]

@flummy1978 Habe an der Hardware nichts verändert und das Binary direkt von Github verwendet. Habe nicht weiter untersucht warum es mit dem selber kompilierten nicht gefunkt hat. Hier mein Pinout:

- CS - D8 (GPIO 15)
- CE - D0 (GPIO 16 - no IRQ!)
- IRQ - D1 (GPIO5)

stefan123t, 2022-11-17:
@Grufbert kannst Du mal die RF24.printPrettyDetails von den beiden Modulen (OK & Fake) einstellen. Wir haben die Vermutung dass sich der Unterschied bei Model Name NRF24L01(+) und RF Data Rate 250kbps zeigen sollte. Danke!

SpeedyC4S, 2022-11-23:
Hallo zusammen, ich habe ebenfalls ein NRF mit ausgefülltem Kreis der funktioniert. Ich nutze die Release 0.5.41 mit den Standardeinstellungen. Ich hatte auch das Problem, dass ich keine Verbindung zum Hoymiles aufbauen konnte. Mir ist dann aber aufgefallen, das die DTU keine ESP-Time hatte, weil ich in der Fritzbox das Internet gesperrt hatte. Nach dem Sync mit dem Browser, wurden auch die Werte nach ca. 2 Minuten  vom Hoymiles angezeigt.
[image]

AdamFerarra, 2022-11-23:
Gerade mal ausprobiert mit der Zeiteinstellung.
Danach funktionierte auch hier das NRF-Modul mit ausgefülltem Kreis sofort.

stefan123t, 2022-11-24:
Hier mal noch ein Bild von einem NRF24L01+ auf einem Modul mit externer Antenne (PA+LNA) geliefert von Makershop / Berrybase das nachweislich funktioniert.

[image]

stefan123t, 2022-11-24:
In Anzeigen oder Beschreibungen taucht auch ab und zu ein Modul mit einem SI24R1 chip von SI labs an.
[image]
> Soweit man das im Netz finden kann, ist der SI24R1 eine schlechte NRF24 Kopie. 
> Die Chinesen nutzen dafür einen gröberen Fertigungs-Prozess, was zu schlechter Reichweite und hohem Energiebedarf führt. 
> Den kannst Du eigentlich nur innerhalb eines Zimmers verwenden.

Laut @cbscpe2 ist das von Makershop gelieferte Modul aber in Wirklichkeit mit einem NRF24L01+ bestückt,
das auch wie erwartet funktioniert, nachdem er nochmal an allen Verbindungen gewackelt und gerüttelt hat.
[image]

hbarthel, 2023-01-29:
Etwas verunsichert wegen des Problems mit den Chips mit Kreis habe ich heute meine AhoyDTU in Betrieb genommen und es funktioniert. Habe 100 µF Elko dran und Zeiteinstellung vorher gemacht.
Habe das hier mit externer Antenne für EUR 4,69: https://www.ebay.de/itm/291793085079