lumapu/ahoy #231: SSID mit Punkten

lalamite, 2022-09-03:
1000 Dank für Eure Mühen.

mich bin von 0.5.5 auf 0.5.15 upgegraded.

Nach dem Formware Update kann sich mein Ahoy nicht mehr mit meinem WLAN verbinden. Mit meinem Gast WLAN funktioniert es allerdings. Der einzige Unterschied zwischen den beiden WLANs ist, dass mein Haupt WLAN Punkte im SSID Namen hat, also so: Wlan…

kann hier etwas schief gegangen sein?

lumapu, 2022-09-04:
muss man prüfen, evtl arbeiten wir hier mit strlen

lumapu, 2022-09-04:
gerade nochmal nachgeschaut, sollte kein Problem sein, es hat nur eine maximale Länge

https://github.com/grindylow/ahoy/blob/main/tools/esp8266/web.cpp#L299

lalamite, 2022-09-04:
> gerade nochmal nachgeschaut, sollte kein Problem sein, es hat nur eine maximale Länge
> 
> https://github.com/grindylow/ahoy/blob/main/tools/esp8266/web.cpp#L299

Hmm strange. Ich habe es 5 Mal probiert. Vorher (0.5.5) ging es aber nun bekomme ich es einfach nicht mehr zum Laufen. Router neu gestartet, lease erneuert etc. nichts hat funktioniert. WLAN Daten von Gastnetzwerk eingetragen und zack…verbunden.

lumapu, 2022-09-04:
das muss nochmal verifiziert werden, habe aktuell keine Punkte in meiner SSID

DanielR92, 2022-09-22:
Seit 18 Tagen gab es keine Rückmeldung. Soll dies noch verfolgt werden?

lumapu, 2022-09-22:
ich würde gerne noch wissen was passiert, hatte es aus den Augen verloren

closed 2022-10-14 by lalamite

stefan123t, 2023-01-12:
@lumapu das könnte mit der überladenen WiFi Methode zum connect zusammenhängen. 
Die interpretiert Angaben mit Punkten vielleicht zuerst als IP Adresse und dann klappt der Fallback nicht. 
Keine Ahnung ob wir das nochmal öffnen müssten oder ob das kein Problem mehr ist ?

lumapu, 2023-01-13:
@stefan123t ich auch nicht 😉