lumapu/ahoy #233: Unterstützung für 3ph HMT-1800-6T, HMT-2250-6T

Tool-use-today, 2022-09-04:
Besten Dank für Eure Entwicklung! die "kleinen" Hoymiles in Fhem einzubinden ist so ziemlich chic gelöst.

Ich bekomme bald zwei HMT-2250-6T, also die dreiphasigen Geräte mit 6 Eingängen (immer 2 Panels auf einem MPPT). 

Wurde schon getested, ob die aktuelle Version auch mit diesen kompatibel ist, bzw. ist es in Planung, diese zu implementieren?

aschiffler, 2022-09-04:
Getestet nein. Wenn es alles gut läuft ist das nur eine Erweiterung im Parser. Wenn Du die Geräte in Betrieb hast sende die Logs der Antworten mit aktiviertem "RF24-Debug". Verwende dazu in den ersten vier Stellen deiner Seriennummer 1161.

Solateur, 2022-09-04:
Quelle: https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#welche-ger%C3%A4te--funktionen-werden-nicht-unterst%C3%BCtzt

"Welche Geräte / Funktionen werden nicht unterstützt

    Werden Hoymiles HMT und HMS auch unterstützt ?

Nein HMT und HMS verwenden ein anderes Funkmodul und können daher (vorerst) auch nicht unterstützt werden. Ob die beiden Modelle auch die selben Kommandos verwenden ist bisher unbekannt."

stefan123t, 2022-09-04:
Eventuell gehen HMT da die evtl nur einen anderen Anschluss haben. aber HMS sind mW definitiv nur mit einem anderen Funkmodul und anderen Frequenzen zu erreichen.

Tool-use-today, 2022-09-05:
Ich habe mir heute die Betriebsanleitung der HMT Serie durchgelesen. Sie benutzt, ebenso wie die HMS Serie die Sub 1G Technologie, wie sie es nennen. Also irgendwas um die 900Mhz.

Ob ich mir die entsprechende DTU-Pro-S, Lite-S dazu kaufe, weiß ich noch nicht. Wenn ja, kann ich das Ding mal aufmachen und prüfen, ob intern die Kommunikation nach dem gleichen Schema läuft und nur ein andere Transmitter genutzt wird.

Evtl. hat das schon mal jemand auch in Bezug auf die HMS Serie gemacht?

stefan123t, 2022-09-07:
HMT-1800-6T HMT-2250-6T datasheets :
https://www.hoymiles.com/wp-content/uploads/2022/08/User-Manual_HMT-18002250-6T_Global_EN_V202207.pdf

Also das T steht hier für Three Phase und es wird wohl offensichtlich ein sog Sub-1G Modul eingesetzt.
Die HMS und auch die dreiphasigen HMT verwenden einen anderen Übertragungsmodus Sub1G
Das ist korrekt, 868MHz soweit ich weiß, HM nutzt 2,4GHz.

> 6. Troubleshooting
This microinverter (SN:1382xxxxxxxx) can only work with the new Hoymiles DTU (DTU-Pro-S, DTU-Lite-S) with the following serial number.

| Model | Serial Number |
| --- | --- |
| DTU-Pro-S | 10FCxxxxxxxx 10FDxxxxxxxx |
| DTU-Lite-S | 10D7xxxxxxxx 10D8xxxxxxxx |

Und hier die DTU-Pro S https://www.shinetech-power.de/wp-content/uploads/2021/07/SNE-DTU-Pro-S.pdf
Hat offenbar auch eine FCC Registrierung mal nach der FCC ID suchen ?

Und hier das Sub1G Module von innen 
https://fcc.report/FCC-ID/2ARNB-HMS10/5204735
<details><summary><b>PARENTAL ADVISORY</b>: Contains Graphical details showing Bare Circuits! 
</summary>
<p>
[image]
[image]
[image]
</p>
</details>

Die Chips haben m.E. folgende Bezeichnungen:
* E230G8 AS2RPK JJ2019 GD ARM (GigaDevices)
* 300A E965 945

Tool-use-today, 2022-09-07:
kann noch das manual mit Pinout dazugeben....

[https://manuals.plus/hoymiles/hms101-sub-1g-transmitter-module-manual#axzz7eCe95wJA](url)

stefan123t, 2022-09-07:
Hier noch unsere bisherigen Erkenntnisse aus dem #hms-serie Kanal im Discord:

Also es gibt keinen HMT-600 und HM-600T ist die Trunk Version des HM-600, kann also auch mit nRF24L01+ gemonitored werden.
Für die HMS-600 und andere aus dieser Baureihe braucht es wohl die o.g. Sub-1G Funkmodule mit Frequenz von 868 MHz oder 915 MHz.

4.1 NRF Channel List
Depending on the program. the module can work on 915MHz for North America and 868MHz for Europe. Unless necessary, it’s forbidden to change the module program.

Read more: https://manuals.plus/hoymiles/hms101-sub-1g-transmitter-module-manual#ixzz7eD7IooaT

stefan123t, 2022-09-07:
Habe noch ein NRF905 Module gefunden https://www.ebay.de/itm/155141668345 dafür gibt es auch eine Arduino Library https://github.com/ZakKemble/nRF905-arduino

There is some information on Zaks Blog: https://blog.zakkemble.net/nrf905-avrarduino-librarydriver/

The nRF905 is a radio transceiver IC similar to the well known nRF24L01, but operates at 433/898/915MHz instead of 2.4GHz, has a much longer range and a few extra IO pins. However, the nRF905 data rate is only 50Kbps compared to nRF24L01’s 2Mbps.

stefan123t, 2022-09-07:
Der von Flole identifizierte CMOSTEK CMT2300A wird hier beschrieben
http://www.cmostek.com/products/wireless/transceivers/CMT2300A

LeoWinterDE, 2022-09-10:
Habe letzte Woche ein HMT-2250-6T in betrieb genommen, sobald das DTU-Pro-S hier ist kann ich auch etwas mitschneiden, sollte bedarf bestehen - ein hackrf liegt hier.

stefan123t, 2022-09-11:
@LeoWinterDE bisher wurden die besten Ergebnisse mit einem/zwei Terminals an den TX/RX Testpunkgen auf dem Board der DTU gewonnen. Die HackRF Auswertung hat sich bei den nRF24L01+ Modulen als schwierig / aufwendig herausgestellt. Zu den CMT2300A haben wir Bezugsquellen über AliExpress / eBay gefunden. Mehr ggf im Discord Chat im Kanal #hms-serie.

jelli123, 2022-09-12:
@LeoWinterDE Ein HackRF Mitschnitt wäre trotzdem interessant, um die Modulationsart zu bestimmen. Wäre doch prima, wenn das auch mit einem klassischen CC1101 ginge.

Flole998, 2022-09-14:
Warum ein CC1101 verwenden wenn's doch die RFM300 gibt und man so quasi die ganze Parametrierung nur mitschneiden muss und durch eine eigene CPU an ein eigenes Funkmodul senden muss? Ich würde erstmal versuchen das ganze vom Grunddesign her nachzubauen, dann kann man am einfachsten bei Problemen nach unterschieden zum Original suchen.

Da ich aber weder Wechselrichter noch Funkmodul habe werde ich dazu leider nicht wirklich mehr beitragen können.

stefan123t, 2022-09-14:
@Flole998 und @jelli123 beide Module CC1101 und RFM300 werden zwar zB auf der Bucht mit 868MHz angeboten. Aber zumindest das CC1101 ist 
1. ungeschirmt
2. i.d.R. ohne Antennenanschluss und nur mit einer festverlöteten Drahtdipolwendel versehen
und zumindest das RFM300 ist
3. viel teurer als ein CMT2300A

@Flole998 wie hast Du eigentlich das CMT2300A initial identifiziert, ich konnte bisher keine Chip IDs etc. zuordnen ?

Das von mir vorgeschlagene nRF905 Modul besitzt lediglich dieselbe Frequenz und fällt daher mE vorerst aus dem Kreis der potentiellen Kandidaten aus.

jelli123, 2022-09-14:
@stefan123t Ob die Schirmung bei 868MHz tatsächlich ausschlaggebend oder eher eine Zulassungsfrage ist, wäre zu untersuchen. Selbst die 2. Oberwelle solle außerhalb des 2.4GHz Bandes liegen. Es gibt aber auch geschirmte CC1101 Modelle.
Bei der Entwicklung auf den Original-Chip zu setzen führt sicher schneller zum Ziel. Mein Gedanke beim CC1101 war, dass es sehr schöne fertige und günstige Geräte zum Sniffen und entwickeln gibt, die man direkt am PC oder RasPi verwenden kann. Siehe die beliebten CUL(W) oder den umflashbaren MaxCube.

Flole998, 2022-09-14:
@stefan123t Die Bezeichnung steht ja auf dem Chip drauf, so habe ich den erkannt.

Es wäre natürlich cool/witzig einen MaxCube als Gateway für diese Wechselrichter umzuflashen, da hat man Ethernet und eigentlich alles was man braucht.

stefan123t, 2022-11-07:
Update von heute aus dem Discord #hms-hmt-serie:

Könnten wir die nRF-Bibliotheken einfach durch CC1101 ersetzen
Ein einfaches austauschen der Libary wird hier nicht wohl ausreichen. 
Die Aufrufe der Libary Funktionen werden nicht identisch sein...
Auch die Befehle für den Wechselrichter werden wahrscheinlich anders sein.

Man braucht also zwingend eine Bibliothek die die Low-Level TX/RX Buffer Kommunikation mit dem CC1101 oder einem anderen 868MHz Chip auf dem genutzten Modul übernimmt. Welche Bibliotheken und Chips haben wir denn eigentlich zur Verfügung ?

Hier der Link zu Texas Instruments CC1101 mit den Application Notes und Beispiel Code etc https://www.ti.com/product/CC1101
Und hier noch eine Bibliothek von Simon Monk basierend auf ElecHouse https://github.com/simonmonk/CC1101_arduino vermutlich ist aber die von SpaceTeddy besser gepflegt / aktueller ? https://github.com/SpaceTeddy/CC1101 
Vielleicht macht es sogar Sinn auf die folgende Universalbibliothek umzustellen dann hätten wir nur ein Interface und könnten sowohl nRF24 als auch CC1101 anbinden / -bieten ? https://github.com/jgromes/RadioLib

Bezüglich den Mitschnitten mit HackRF etc. das sollte eigentlich auch mit einem RTL-SDR gehen, da ja das 868MHz Band weit unter der oberen ~2.0GHz Grenzfrequenz des RTL-SDR liegt. Nur bei den 2.4GHz des NRF24 war ein HackRF notwendig.

Ich weiß nicht ob die Schrauben der DTU Pro S irgendwie mit Aufklebern versiegelt sind?

Aber die Testpunkte für TX/RX zwischen STM32 MCU und NRF24 waren bisher klar gekennzeichnet und eine/zwei USB2Serial Converter sind leicht angeschlossen damit kann man die Kommunikation quasi im Klartext mit einem Terminal mitschneiden. HackRF / RTL-SDR zu dekodieren ist da weit komplizierter und vermutlich nicht zwingend notwendig.

FFW-Scripter, 2022-11-15:
Hier mal etwas Bildmaterial dazu
[image]
[image]
[image]
[image]
[image]
[image]

FFW-Scripter, 2022-11-15:
In Bild drei sind recht Eindeutig die Messpunkte für RX/TX sichtbar (FT-TX3/FT-RX3)

Argafal, 2022-11-24:
@FFW-Scripter Was zeigen denn die Bilder? Eine Hoymiles DTU Pro von innen, die Version für HMS und HMT? Ist das deiner oder hast du die Bilder irgendwo online gefunden?

Ich hab gerade zwei HMS-2200-4T gekauft und verfolge diesen Issue jetzt auch mit großer Spannung.

FFW-Scripter, 2022-11-24:
Ja, das ist die DTU Pro S von innen. Mittlerweile hat @lumapu diese und noch eine weitere funktionierende zum mitschneiden.

karrenbauer, 2022-11-28:
ich habe auch einen HMT-2250-6T und verfolge das Thema mit Spannung....

lumapu, 2022-11-28:
in Discord drive schon mehr Details bekannt

VincentSC, 2022-12-09:
Can you share a master-plan on [wiki](https://github.com/lumapu/ahoy/wiki/] once it's shared on Discord? I only see people buy hardware from Amazon, and have no idea if I can help anywhere

mbay0r, 2022-12-15:
will this also include hms-2000-4T ?

Budd1900, 2022-12-16:
Hallo, ich verfolge das Thema ebenso mit Spannung und könnte ggf auch aktiv oder auch nur mit ein paar Euro für Kaffee unterstützen. Wie kann ich im Discord beitreten bzw kann mir bitte jemand eine Einladung zukommen lassen? Wie können wir helfen?

karrenbauer, 2022-12-16:
https://discord.com/channels/984173303147155506/993500210271629393

karrenbauer, 2022-12-16:
einfach bei discord registrieren:
https://discord.com/channels/984173303147155506/993500210271629393

Tobias007, 2023-01-01:
Hi, würde gerne bei Discord mitlesen - der Link oben geht allerdings nicht …

lumapu, 2023-01-01:
das ist der richtige Link:
https://discord.gg/WzhxEY62mB

biasowicka, 2023-03-03:
Czy już działa na HMT-2250 ?

lumapu, 2023-03-06:
@biasowicka can you write in English?

FFW-Scripter, 2023-03-07:
> Czy już działa na HMT-2250 ?

EN: Not at the moment, @lumapu and others are wirking on it ;-)
PL: W tej chwili nie, @lumapu i inni nad tym pracują ;-)

MiniOh, 2023-04-13:
Hallo,

gibt es bereits Pläne, wann die HMS Serie supportet werden könnte?

PawelSpoon, 2023-04-23:
one should first read then buy :)
just received and started my HMT-1800-6T
what a surprise that its not supported yet ?

lumapu, 2023-04-25:
> Hallo,
> 
> gibt es bereits Pläne, wann die HMS Serie supportet werden könnte?

Ich habe eine ESP32 Version, ich kann sie gerne später als weiteren Branch veröffentlichen. HMS läuft aktuell nur auf dem ESP32

0x7878, 2023-05-09:
@lumapu würdest du deinen aktuellen Stand Veröffentlichen? Vielleicht kann ich unterstützen für den esp8266.

lumapu, 2023-05-09:
bitte in #904 schauen, da ist auch schon eine Firmware, ansonsten im Branch `hms`.

lumapu, 2023-05-25:
wir diskutieren in #904 weiter

closed 2023-05-25 by lumapu

dnwalker, 2023-09-10:
Hallo,

wird auch der HMS-1800-4T unterstützt?

Der HMS-1800 ist ja aufgeführt aber es wäre schön wenn man eine sicherer Bestätigung für den HMS-1800-4T hätte-

Danke und Gruß
Dieter