lumapu/ahoy #234: 0.5.16 Anzeige Leistungslimitierung

hubsi5, 2022-09-04:
habe die nagelneue Version raufgespielt.
Bei mir wird in der Visualisierung nur mehr 100% Limit angezeigt.
Ich limitiere per MQTT die Leistung mit absoluter Leistungsangabe.
Die Limitierung funktioniert auch, nur wir dann noch immer 100% angezeigt.
In der 0.5.15 Version funktionierte das noch einwandfrei.

Chrischi-, 2022-09-04:
siehe meinen Kommentar https://github.com/grindylow/ahoy/issues/228#issuecomment-1236277850

Dachte mir schon, dass es ein Bug ist.

hubsi5, 2022-09-04:
habe nun auch ERase Settings durchgeführt. Aber der Fehler ist bei mir noch immer vorhanden

aschiffler, 2022-09-04:
Hi,

der angezeigte Wert war vorher Version < 0.5.16 immer eine Kopie des Sollwerts. Nun ist es so, dass der angezeigte Wert der Istwert ist der im WR eingestellt ist. Dieser Wert wird immer in Prozent angegeben.

closed 2022-09-04 by aschiffler

hubsi5, 2022-09-04:
Es geht bei mir trotzdem nicht.
Wenn ich meinen 1500er auf 600W limitiere steht im ahoy noch immer 100%

Chrischi-, 2022-09-04:
@hubsi5 
es kann natürlich sein, dass Ahoy die Änderungen nicht in den WR geschrieben hat. In meinem Fall evtl. nach den ganzen Resets und Neustarts dann aber irgendwann erfolgreich übertragen hat.

@aschiffler 
Gibt es irgendwo die Möglichkeit, das „Antwort-OK“ (erfolgreich in den WR geschrieben) zu sehen?

hubsi5, 2022-09-04:
Der Wechselrichter limitiert ja, nur die Anzeige bleibt bei 100%