lumapu/ahoy #237: new version 5.16, "..ahoy-main/tools/esp8266/html/h" is missing

duenengeist, 2022-09-04:
Hi all,

I downloaded the zip for 5.16 and tried to compile. The content of "ahoy-main/tools/esp8266/html/h" is missing.
I copied the folder from 5.15, that made it compile. But the might have been some changes on those files ?

Maybe someone can have al ook?

cheers,

Carsten

aschiffler, 2022-09-04:
Hi,

as you can see here
https://github.com/grindylow/ahoy/blob/01481084a60dccbde9a28723808a9f8d392ff5ca/tools/esp8266/platformio.ini#L32
during the build proccess the content in ../h is created by a python script. So the issue is within your build environment.
You can use of course the automatic build firmware under the release section.

closed 2022-09-04 by aschiffler

duenengeist, 2022-09-04:
Sorry, das ich nochmal zurückfrage: bisher habe ich immer die Arduinio-Umgebung zum Übersetzen benutzt. Das heisst dann das geht jetzt nicht mehr? Das wäre schade.

"You can use of course the automatic build firmware under the release section" --> wo finde ich denn die?

lumapu, 2022-09-04:
hier in Github unter Actions, im die development builds zu laden muss man angemeldet sein 

Es geht auch weiterhin mit Arduino IDE, allerdings muss man zuerst das python script in html Ordner manuell ausführen

duenengeist, 2022-09-04:
merci, hab's hinbekommen. Die 5.16 läuft bei mir bisher ohne Aufälligkeiten :-)