lumapu/ahoy #239: Visualisierung mal mit Daten mal ohne Version 0.5.16

stevedee78, 2022-09-04:
Hallo, habe gerade die 0.5.16 über die update Funktion eingespielt und ein Problem (siehe Video) fesgestellt. Unteranderem wechselt die Meldung "Inverter #0: HM-400 (v1) is available and is producing" zu "Inverter #0: HM-400 (v1) is available and is not producing" und zurück.

https://user-images.githubusercontent.com/25169154/188318552-4d203a86-9bc3-4aba-a20d-bdd67344a309.mp4

Git SHA: g7f01588

Pony1966, 2022-09-04:
Beim HM600 auch

DennisXK, 2022-09-04:
Wie weit ist die DTU vom WR entfernet?

Pony1966, 2022-09-04:
3 Meter, wie bei der 0.5.15 die ja lief

stevedee78, 2022-09-04:
> Wie weit ist die DTU vom WR entfernet? 

3-4m. Mit der 0.5.15 gab es keine Probleme.

DennisXK, 2022-09-04:
Ok, weil mir ist das auch schon aufgefallen. Dachte es liegt an der Entfernung von ca 10m

aschiffler, 2022-09-04:
Wie oft passiert das hintereinander. Ich vermute mit einmal.
Oder dauerhaft?

stevedee78, 2022-09-04:
> Wie oft passiert das hintereinander. Ich vermute nur einmal.
> Oder dauerhaft?

Bei mir dauerhaft.

Pony1966, 2022-09-04:
Dauerhaft , kleinster Intervall ca 7 sec pro Prod oder Prod Not.

lumapu, 2022-09-04:
auf was steht im Setup die Auswahl für das powerLimit?

Pony1966, 2022-09-04:
Absolute in Watt non persistent

DennisXK, 2022-09-04:
Bei mir steht es auf „no power limit“

stevedee78, 2022-09-04:
> auf was steht im Setup die Auswahl für das powerLimit?

Noch power limit

stefan123t, 2022-09-04:
Sieht aus als wäre keine komplette Payload zurückgekommen. Ist aber nur eine Vermutung da dann ja evtl die aktuellen Werte überschrieben oder auf 0 gesetzt werden ?

gandalfred, 2022-09-05:
Bei mir das gleiche. Wenn ich mehrmals die IP des Ahoy mit 0.5.16 anfrage kommt die verschwundene Anzeige dann wieder, bis sie dann nach weiteren Anfragen abermals wieder verschwindet. Mit 0.5.15 läuft das ohne Probleme.

aschiffler, 2022-09-05:
Hi,
wurde nun gelöst. Es war ein Fehler im Parser für die einkanaligen WR. Ein Abgleich mit dem Projekt OpenDTU hat dies auch nochmals bestätigt.
Die letzte Alarm Message ID wurde falsch erfasst was dazu führt, dass ahoy "denkt" es liegt ein neuer Alarm vor und ruft diesen ab. Während dieses Abrufs belibt die Web-Anzeige leer. Somit passiert  dies zyklisch wenn der Alarm nicht korrekt gelesen werden kann vom WR.

Bitte nun mit dem dev-build testen.

Siehe:
https://github.com/grindylow/ahoy/commit/03a6b87cc1d889d6082c0b4e088d530f8990f914

closed 2022-09-05 by aschiffler