lumapu/ahoy #246: not producing / producing 0.5.17

Pony1966, 2022-09-05:
Die Anzeige springt zwischen producing und not producing hin und her.
Mqtt Datenpunkte werden im Mqtt Broker nicht mehr erstellt.
Ein Wechsel zwischen No Limit und Absolut in Watt non Persistent zeigt keine Veränderung.
Payload unterschiedlicher Länge sind per hterm zu sehen.
Wemos D1, HM600, Soft 0ae56ef

aschiffler, 2022-09-05:
Bitte mal ca. 20 Sekunden serial log dazu posten.

Pony1966, 2022-09-05:
Log
`I: connect to network '500' ...<\r><\n>
...........<\r><\n>
........................................<\r><\n>
........................................<\r><\n>
....................................<\r><\n>
I: <\n><\n>----------------------------------------<\r><\n>
I: Welcome to AHOY!<\r><\n>
I: <\n>point your browser to http://192.168.2.00<\r><\n>
I: to configure your device<\r><\n>
I: ----------------------------------------<\n><\r><\n>
I: RF24 Amp Pwr: RF24_PA_I: HIGH<\r><\n>
I: Radio Config:<\r><\n>
SPI Frequency<9><9>= 1 Mhz<\n>Channel<9><9><9>= 3 (~ 2403 MHz)<\r><\n>
Model<9><9><9>= nRF24L01+<\r><\n>
RF Data Rate<9><9>= 250 KBPS<\r><\n>
RF Power Amplifier<9>= PA_HIGH<\r><\n>
RF Low Noise Amplifier<9>= Enabled<\r><\n>
CRC Length<9><9>= 16 bits<\r><\n>
Address Length<9><9>= 5 bytes<\r><\n>
Static Payload Length<9>= 32 bytes<\r><\n>
Auto Retry Delay<9>= 250 microseconds<\r><\n>
Auto Retry Attempts<9>= 0 maximum<\r><\n>
Packets lost on<\n>    current channel<9>= 0<\r><\n>
Retry attempts made for<\n>    last transmission<9>= 0<\r><\n>
Multicast<9><9>= Disabled<\r><\n>
Custom ACK Payload<9>= Disabled<\r><\n>
Dynamic Payloads<9>= Enabled<\r><\n>
Auto Acknowledgment<9>= Disabled<\r><\n>
Primary Mode<9><9>= RX<\r><\n>
TX address<9><9>= 0xe7e7e7e7e7<\r><\n>
pipe 0 (closed) bound<9>= 0xe7e7e7e7e7<\r><\n>
pipe 1 ( open ) bound<9>= 0x1234567801<\r><\n>
pipe 2 (closed) bound<9>= 0xc3<\r><\n>
pipe 3 (closed) bound<9>= 0xc4<\r><\n>
pipe 4 (closed) bound<9>= 0xc5<\r><\n>
pipe 5 (closed) bound<9>= 0xc6<\r><\n>
I: [NTP]: 2022-09-05 18:44:28<\r><\n>
I: enqueued cmd failed/timeout<\r><\n>
I: Inverter #0 I: no Payload received! (retransmits: 0)<\r><\n>
I: Requesting Inverter SN 114181867721<\r><\n>
I: RX 27B Ch23 | 95 81 86 77 21 81 86 77 21 81 27 1A 07 E5 04 4D 03 4A 00 68 00 00 00 00 E6 FB BE <\r><\n>
I: RX 27B Ch3 | 95 81 86 77 21 81 86 77 21 01 00 01 00 F2 00 28 00 62 01 F4 00 33 01 00 00 00 EA <\r><\n>
I: RX 23B Ch3 | 95 81 86 77 21 81 86 77 21 83 00 00 00 0E 03 E8 01 29 24 8B 2C 68 30 <\r><\n>
E: while retrieving data: Frame 2 missing: Request Retransmit<\r><\n>
I: RX 27B Ch3 | 95 81 86 77 21 81 86 77 21 02 5D 90 00 00 17 DA 07 6F 11 4C 09 32 13 85 01 52 5C <\r><\n>
I: RX 27B Ch23 | 95 81 86 77 21 81 86 77 21 02 5D 90 00 00 17 DA 07 6F 11 4C 09 32 13 85 01 52 5C <\r><\n>
I: Payload (42): 00 01 00 F2 00 28 00 62 01 F4 00 33 01 00 00 00 5D 90 00 00 17 DA 07 6F 11 4C 09 32 13 85 01 52 00 00 00 0E 03 E8 01 29 24 8B <\r><\n>
I: Inverter: 0<\r><\n>
I: myInv0/ch0/FWVersion: 1.000 <\r><\n>
I: myInv0/ch0/FWBuildYear: 242.000 <\r><\n>
I: myInv0/ch0/FWBuildMonthDay: 40.000 <\r><\n>
I: myInv0/ch0/HWPartId: 500.000 <\r><\n>
I: <\r><\n>
I: Inverter: 0<\r><\n>
I: myInv0/ch0/FWVersion: 1.000 <\r><\n>
I: myInv0/ch0/FWBuildYear: 242.000 <\r><\n>
I: myInv0/ch0/FWBuildMonthDay: 40.000 <\r><\n>
I: myInv0/ch0/HWPartId: 500.000 <\r><\n>
I: <\r><\n>
I: Inverter: 0<\r><\n>
I: myInv0/ch0/FWVersion: 1.000 <\r><\n>
I: myInv0/ch0/FWBuildYear: 242.000 <\r><\n>
I: myInv0/ch0/FWBuildMonthDay: 40.000 <\r><\n>
I: myInv0/ch0/HWPartId: 500.000 <\r><\n>
I: <\r><\n>
I: Requesting Inverter SN 114181867721<\r><\n>
I: RX 27B Ch3 | 95 81 86 77 21 81 86 77 21 81 00 01 00 53 00 00 03 E8 FF FF FF FF 01 68 9F 8F D4 <\r><\n>
I: RX 27B Ch23 | 95 81 86 77 21 81 86 77 21 81 00 01 00 53 00 00 03 E8 FF FF FF FF 01 68 9F 8F D4 <\r><\n>
I: RX 27B Ch3 | 95 81 86 77 21 81 86 77 21 01 00 01 00 E7 00 2A 00 61 01 F4 00 33 01 00 00 00 FE <\r><\n>
[... 132 lines cut ...]
I: Inverter: 0<\r><\n>
I: myInv0/ch1/U_DC: 23.900 V<\r><\n>
I: myInv0/ch1/I_DC: 0.410 A<\r><\n>
I: myInv0/ch1/P_DC: 9.800 W<\r><\n>
I: myInv0/ch1/YieldDay: 1903.000 <\r><\n>
I: myInv0/ch1/YieldTotal: 23.952 kW<\r><\n>
I: myInv0/ch1/Irradiation: 2.882 %<\r><\n>
I: myInv0/ch2/U_DC: 50.100 V<\r><\n>
I: myInv0/ch2/I_DC: 0.480 A<\r><\n>
I: myInv0/ch2/P_DC: 24.100 W<\r><\n>
I: myInv0/ch2/YieldDay: 4429.000 <\r><\n>
I: myInv0/ch2/YieldTotal: 6.107 kWh<\r><\n>
I: myInv0/ch2/Irradiation: 7.088 %<\r><\n>
I: myInv0/ch0/U_AC: 235.300 V<\r><\n>
I: myInv0/ch0/I_AC: 0.140 A<\r><\n>
I: myInv0/ch0/P_AC: 32.400 W<\r><\n>
I: myInv0/ch0/Freq: 49.980 Hz<\r><\n>
I: myInv0/ch0/Temp: 29.700 Â°<\r><\n>
I: myInv0/ch0/ALARM_MES_ID: 9363.000 <\r><\n>
I: myInv0/ch0/YieldDay: 6332.000 <\r><\n>
I: myInv0/ch0/YieldTotal: 30.059 kW<\r><\n>
I: myInv0/ch0/P_DC: 33.900 W<\r><\n>
I: myInv0/ch0/Efficiency: 95.575 %<\r><\n>
I: <\r><\n>
`

Pony1966, 2022-09-05:
not Producing wenn 2 Alarm Code kommt.

[image]

DirkausHH, 2022-09-06:
Bei meinem HM-300 und 0.5.17 passiert das nicht.

aschiffler, 2022-09-06:
Ok. Der Code oben ist ab reboot/start von ahoy. Hier kommt es bedingt durch die Abfrage von Firmware / Alarm Code zu Zuständen die nicht eindeutig sind.
Siehe dazu die Doku hier: https://github.com/grindylow/ahoy/blob/main/tools/esp8266/User_Manual.md#developer-information-about-command-queue
Es sollte aber nachdem "nur noch" das enqueuedCmd: 11 kommt funktionieren.
Im aktuellen Release müsste das so funktionieren.

Bitte nochmals berichten / testen und ggf. issue wieder öffnen

closed 2022-09-06 by aschiffler