lumapu/ahoy #251: Feature Request: Secure Connection Ahoy<->MQTT

c1328, 2022-09-06:
Right know there are only unsecured connections between Ahoy and MQTT possible. If your MQTT broker is outside of your LAN/WLAN there is a high security risk.

stefan123t, 2022-09-07:
@c1328 while this (TLS secure transport) certainly should be on the TODO list we have not planned this feature yet.
We are focusing on the AhoyDTU protocol details with the Inverter and especially the switch from ESPWebServer -> ESPAsyncWebServer at the moment.
So there are no free resources to implement TLS at the moment, but PR are welcome.

madmartin, 2022-09-09:
@c1328 There is a sister project [OpenDTU](https://github.com/tbnobody/OpenDTU) which has MQTT w/ TLS implemented.

stefan123t, 2022-09-09:
So maybe after we have implemented AsyncWeb03 branch someone may add TLS based on the OpenDTU feature. I guess this should be possible being on the same libraries then.