lumapu/ahoy #252: DNS issues since v05.15

fubu2k, 2022-09-06:
since v05.15 my PiHole is reporting DNS issues. An unknow (NaN) error ist reported every 5 seconds (see screenshot). 

[image]

stefan123t, 2022-09-07:
I do see this repeats quite frequently and it may be related to the NTP / MQTT broker name resolution.
Could you try it with a later version 0.5.17 which should have these fixed ?

fubu2k, 2022-09-07:
hello stefan, i tried version ahoy_v0.5.18_dev_build and the problem still exists.

stefan123t, 2022-09-07:
Can you try to enable the Debug mode and figure out when/where it is issuing these queries according to the logfiles?

fubu2k, 2022-09-07:
ahoy-dtu is the dns name for the controller - setup in the ahoy-gui. 
.iot is the domain, which is pulled to the controller via dhcp server. 
nslookup is resolving the dns name ahoy-dtu.iot successfully, but the unknow "NaN-Error" (not a number) still exists. 

iam trying to debug the pihole dns logfiles

aschiffler, 2022-09-08:
That's not an issues of ahoy esp8266. I guess that the IP/DNS name of the MQTT broker is not set correct so the DNS query fails.

fubu2k, 2022-09-08:
after debugging pihole log files in debug-mode dns is running fine. there are no errors in the log files. 
iam not using mqtt broker. the configuration in the setup menu is empty. maybe thats the problem?

aschiffler, 2022-09-08:
Maybe "empty" is not "nothing". try to re-setup the ahoy by clicking erase-flash.

fubu2k, 2022-09-08:
i fixed it by using 127.0.0.1 localhost ip for MQTT broker. 
After that i re-setup the ahoy. 
problem is fixed.

seems to be an configuration problem with a firmware update.

thanks :)

closed 2022-09-08 by aschiffler