lumapu/ahoy #258: Support für MI-300+/MI-600+ & MI-1200+ Gen2 Inverter mit 0x08/0x09, 0x11/0x12 bzw. 0x36..0x39 Kommandos

stefan123t, 2022-09-08:
Die Gen2 MI Inverter verwenden ein anderes Kommando ( 0x08/0x09, 0x11/0x12 bzw. 0x36..0x39 ) um Daten und Events vom Wechselrichter zu bekommen.
Dafür müssen Anpassungen an hmRadio.h und app.cpp vorgenommen werden um diese älteren Wechselrichter der 2ten Generation ebenfalls in AhoyDTU zu unterstützen.

stefan123t, 2022-09-08:
Die Anfrage Pakete für die Informationen (DevInfo 0x15) werden mit sendTimePacket() erstellt und gesendet und mW in app::loop() der app.cpp in mehreren Paketen empfangen (hier braucht man nur eine Anfrage und es kommen N Antwort-Pakete) und dann die Payload mit processPayload(true) geparst.
https://github.com/grindylow/ahoy/blob/5402e9b0a2cf1c80012dcf62d89179693abc15b5/tools/esp8266/app.cpp#L74-L117

bzw.

https://github.com/grindylow/ahoy/blob/5402e9b0a2cf1c80012dcf62d89179693abc15b5/tools/esp8266/app.cpp#L155-L158

aschiffler, 2022-09-08:
So ist es. Wobei das Verarbeiten der Antwort auch für nur ein Antwort Frame jetzt schon funktioniert. So wie andere "HM" Kommandos ja auch nur ein Antwort Frame haben können.

rejoe2, 2022-09-08:
Achtung: Auf eine 0x09-Anfrage kommen ZWEI Antworten, es gibt keine separate Anfrage für0x88, und entsprechen kommen auf 0x11 dann 0x91 und 0x92 als Antwort.
Die Antworten liegen aber auf unterschiedlichen Kanälen.

stefan123t, 2022-09-08:
@rejoe2 

die Modelle werden im AhoyDTU in hmSystem.h ausgewertet und anhand der IDs unterschieden:
https://github.com/grindylow/ahoy/blob/348ad4bd07ddde67cd017547b2afcafa486f2b09/tools/esp8266/hmSystem.h#L48-L57

rejoe2, 2022-09-08:
Da hatte ich es geklaut und in die setup() wie folgt integriert:
````
uint64_t sn = SerialWR;
  longlongasbytes llsn;
  llsn.ull = sn;
  if(llsn.bytes[5] == 0x10) {
  switch(llsn.bytes[4]) {
    case 0x21: 
        { MI300 = 1; MI600 = 0; MI1500 = 0; 
        if (NRofPV==0) { NRofPV = 1; } 
        break; }
    case 0x41: 
        { MI600 = 1; MI300 = 0; MI1500 = 0; 
        if (NRofPV==0) { NRofPV = 2; } 
        break; }
    case 0x61: 
        { MI1500 = 1; MI600 = 0; MI300 = 0; 
        if (NRofPV==0) { NRofPV = 4; }
        break; }
  }
  MAXPOWER = NRofPV * PVPOWER;   
  MINPOWER = int(MAXPOWER / 10);
  }
  ````

stefan123t, 2022-09-08:
@lumapu Vielleicht INV_TYPE_1CH in HM_TYPE_1CH bzw. MI_TYPE_1CH unterscheiden oder fällt uns das dann bei der Darstellung wieder auf die Füße ?

Aber besser wäre es wahrscheinlich ein p->model = INV_GEN2_MI bzw. INV_GEN3_HM einzuführen ?
Dann kann man weiterhin die Darstellung mit INV_TYPE_XCH machen und an anderer Stelle (Senden & Payload Parser) einfach auf die INV_GEN2_MI / INV_GEN3_HM zugreifen.

lumapu, 2022-09-08:
umbenennen geht immer, beide Varianten funktionieren. Die Erfahrung zeigt, dass man erst beim implementieren feststellt, welche besser / sauberer ist

Ziyatoe, 2022-09-12:
> Aber besser wäre es wahrscheinlich ein p->gen = INV_GEN2_MI bzw. INV_GEN3_HM einzuführen ? Dann kann man weiterhin die Darstellung mit INV_TYPE_XCH machen und an anderer Stelle (Senden & Payload Parser) einfach auf die INV_GEN2_MI / INV_GEN3_HM zugreifen.

man muss ja beim rx-parsen auch unterscheiden können, darum braucht es INV_GEN2_MI oder?

stefan123t, 2022-11-07:
@Ziyatoe wie heute im Discord #ahoy-esp8266 besprochen wäre es notwendig bei den Kommandos zwischen Gen2 und Gen3 zu unterscheiden.
* Gen2 : ein oder mehrere Cmd Request Paket(e) und je ein Cmd Response Paket. Keine Paket ID
* Gen3 : ein Cmd Request Paket und ein oder mehrere Cmd Response Pakete mit Paket ID.

Die Daten gehen dann evtl. auch über verschiedene Backends nrf24Radio.h bzw. cc1101Radio.h an die Wechselrichter raus.

closed 2023-05-25 by lumapu