lumapu/ahoy #259: Irritating Tag "refs/heads/main" on this repository

madmartin, 2022-09-08:
This repository currently contains the tags 

```
ahoy_v0.5.16
ahoy_v0.5.17
refs/heads/main

```
The last tag `refs/heads/main` is most probably a mistake and it looks like its name should be something like `v0.5.15`.

Additionally, this tag name confuses git and causes an error messages when I try to push in my forked repository:

```
git push
error: dst refspec refs/heads/main matches more than one
error: failed to push some refs to 'github.com:madmartin/ahoy.git'
```

Please delete this tag.

lumapu, 2022-09-08:
I came across this annoying tag done days ago, but it wasn't possible for me to delete it. 
There are several people around the world having the same problem - hopefully I find one working fix for that.

I don't know if the tag will be there again if someone pushes it to the Repository again ...

lumapu, 2022-09-08:
@madmartin fixed this issue, please try again.

my commands were:

```
git tag -d refs/heads/main
git push origin :refs/tags/refs/heads/main
```

madmartin, 2022-09-09:
Hello @lumapu ,

I fixed it with a slightly different command set, but managed to fix it for my fork, too.

Your expectation to re-appear this tag is right,  so everyone who has write access to `https://github.com/grindylow/ahoy` must also execute 

`git tag -d refs/heads/main`

on his/her local copy of the repository.

Maybe you have a list of these people and their accounts and can notify them by mentioning the account names in a message in this issue.

lumapu, 2022-09-09:
@aschiffler @stefan123t bitte löscht bei euch doch lokal auch das main Tag

aschiffler, 2022-09-09:
I delete the tag in local copy. But I have to do a commit with 2 added files for the ESP32 schematic (?),

closed 2022-09-09 by aschiffler

madmartin, 2022-09-09:
@aschiffler strange.... now my additions to README.md from  #261 are gone, the two images `doc/Wiring_ESP32_*.png` exist. My commit 2db984f1c049120e15c603859e315f48e2fb6b96 is gone.

lumapu, 2022-09-09:
sorry for my stupid question: are you pointing to the correct Branch 'main'?

madmartin, 2022-09-09:
Yes, I point to "main", but I still have the branches for the pull requests available, and IMHO there are still commits in the object store from deleted branches. 

I found the commit number by using `gitg`, see screenshot below.

I think @aschiffler has overwritten commit 2db984f1c049120e15c603859e315f48e2fb6b96 with new commit 044771b75da93c571b80ef7f1579625bb0011068 with a force-push.

```
martin@bln9716cm ~/swbuild/ahoy/tools/esp8266 $ git branch 
  fix-readme
* main
  userman
  userman-add-inverter
martin@bln9716cm ~/swbuild/ahoy/tools/esp8266 $ git show 2db984f1c049120e15c603859e315f48e2fb6b96
commit 2db984f1c049120e15c603859e315f48e2fb6b96
Author: Martin Dummer <martin.dummer@gmx.net>
Date:   Thu Sep 8 13:06:26 2022 +0200

    esp8266/README.md: add info and wiring for ESP32 variant
    
    Signed-off-by: Martin Dummer <martin.dummer@gmx.net>

diff --git a/doc/Wiring_ESP32_Schematic.png b/doc/Wiring_ESP32_Schematic.png
new file mode 100644
index 0000000..e33e0c8
Binary files /dev/null and b/doc/Wiring_ESP32_Schematic.png differ
diff --git a/doc/Wiring_ESP32_Symbol.png b/doc/Wiring_ESP32_Symbol.png
new file mode 100644
index 0000000..0458439
Binary files /dev/null and b/doc/Wiring_ESP32_Symbol.png differ
diff --git a/tools/esp8266/README.md b/tools/esp8266/README.md
index a462843..f2fa10d 100644
--- a/tools/esp8266/README.md
+++ b/tools/esp8266/README.md
@@ -76,11 +76,29 @@ Additional, there are 3 pins, which can be set individual:
 
 #### ESP8266 wiring example
 This is an example wiring using a Wemos D1 mini.<br>
-<img src="https://github.com/grindylow/ahoy/blob/main/doc/AhoyWemos_Steckplatine.jpg" width="300">
-<img src="https://github.com/grindylow/ahoy/blob/main/doc/AhoyWemos_Schaltplan.jpg" width="300">
+##### Schematic
+![Schematic](../../doc/AhoyWemos_Schaltplan.jpg)
 
+##### Symbolic view
+![Symbolic](../../doc/AhoyWemos_Steckplatine.jpg)
 
+#### ESP32 wiring example
+Example wiring for a 38pin ESP32 module
 
+##### Schematic
+![Schematic](../../doc/Wiring_ESP32_Schematic.png)
+
+##### Symbolic view
+![Symbolic](../../doc/Wiring_ESP32_Symbol.png)
+
+##### ESP32 GPIO settings
+For this wiring, set the 3 individual GPIOs under the /setup URL:
+
+```
+CS   D1 (GPIO5)
+CE   D2 (GPIO4)
+IRQ  D0 (GPIO16 - no IRQ!)
+```
 
 ## Flash the Firmware on your Ahoy DTU Hardware
 Once your Hardware is ready to run, you need to flash the Ahoy DTU Firmware to your Board.
martin@bln9716cm ~/swbuild/ahoy/tools/esp8266 $ 
```

[image]

lumapu, 2022-09-09:
now your changes should be there again

aschiffler, 2022-09-09:
I did not force the push.
As i have written it was also Strange for me.
I hope nothing unreversable happend.