lumapu/ahoy #262: Startverhalten ab v0.5.17 langsamer

bluespiano, 2022-09-08:
Danke erstmal an alle Beteiligten für die tolle Arbeit!
V0.5.17
Verhalten: Nach Reboot dauert der Start mit der .17 ungewöhnlich lange - habe mal die Statusänderungen (available / not available / producing / not producing) mit Zeitstempel versehen kopiert.  Vielleicht hilft es, das Startverhalten zu optimieren. 

>**Reboot**
Uptime: 0 Days, **00:00:17**; now: 2022-09-08 18:03:17
Statistics:
Receive success: 1
Receive fail: 0
Frames received: 8
Send Cnt: 2
Inverter #0: HM1500_Anbau (v0) is not available and is not producing
Inverter #1: HM1500_Werkst (v0) is available and is producing
Inverter #2: n/a
MQTT: connected
++++++++++++++
Uptime: 0 Days, **00:00:47**; now: 2022-09-08 18:03:47
Statistics:
Receive success: 2
Receive fail: 0
Frames received: 14
Send Cnt: 4
Inverter #0: HM1500_Anbau (v0) is available and is producing
Inverter #1: HM1500_Werkst (v0) is available and is producing
Inverter #2: n/a
MQTT: connected
++++++++++++++
Uptime: 0 Days, **00:01:21**; now: 2022-09-08 18:04:21
Statistics:
Receive success: 3
Receive fail: 0
Frames received: 16
Send Cnt: 5
Inverter #0: HM1500_Anbau (v0) is available and is producing
Inverter #1: HM1500_Werkst (v10018) is available and is not producing
Inverter #2: n/a
MQTT: connected
++++++++++++++++
Uptime: 0 Days, **00:01:49**; now: 2022-09-08 18:04:49
Statistics:
Receive success: 4
Receive fail: 0
Frames received: 18
Send Cnt: 7
Inverter #0: HM1500_Anbau (v10018) is available and is not producing
Inverter #1: HM1500_Werkst (v10018) is available and is not producing
Inverter #2: n/a
MQTT: connected
+++++++++++
Uptime: 0 Days, **00:04:17**; now: 2022-09-08 18:07:17
Statistics:
Receive success: 9
Receive fail: 0
Frames received: 33
Send Cnt: 14
Inverter #0: HM1500_Anbau (v10018) is available and is not producing
Inverter #1: HM1500_Werkst (v10018) is available and is producing
Inverter #2: n/a
MQTT: connected
++++++++++++
Uptime: 0 Days, **00:04:56**; now: 2022-09-08 18:07:56
Statistics:
Receive success: 10
Receive fail: 0
Frames received: 41
Send Cnt: 16
Inverter #0: HM1500_Anbau (v10018) is available and is producing
Inverter #1: HM1500_Werkst (v10018) is available and is producing
Inverter #2: n/a
MQTT: connected

stefan123t, 2022-09-08:
Er zeigt keine Fehler an: Receive fail: 0 
Man kann sehen dass er die Firmware version der Inverter abfragt und anzeigt: (v0) -> (v10018)
Mit der entsprechenden Debug Option sieht man mE. auch was er an den WR sendet / empfängt.

lumapu, 2022-09-18:
in asyncWeb03 wurde es verbessert, bitte damit nochmal prüfen

DanielR92, 2022-10-13:
dieses Issue ist seit 25 Tagen schon offen und es kam keine Rückmeldung mehr. 
Ich konnte diesen Fehler nicht mehr beobachten. Könnte man diesen dann closen?

closed 2022-10-14 by lumapu