lumapu/ahoy #267: Deaktiviere ungenutzte Peripherie um Strom zu sparen

thfcm, 2022-09-11:
Vorschlag:

Um die 3,3V-Schiene zu entlasten, könnte man im Mikrocontroller einige Peripherie deaktivieren.
Die Einsparungen müssten noch ermittelt werden (via Test oder eventuell gibt es bereits was im Netz).

Die Einsparungen könnten ggf. den Spannungsregler soweit entlasten, dass mehr Reserve für die Lastspitzen des nRF-Moduls verfügbar wären.

Beim **ESP32** geht das angeblich, in dem man die einzelnen Teile vom Clocking abkoppelt.

Siehe [ESP32-Manual ](https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) Punkt _5.3.7 Peripheral Clock Gating and Reset_ sowie auf den darauf folgenden Seiten, z.B. 
 *  Seite 97 _DPORT_PERI_RST_EN_REG (0x020)_ (Deaktivieren von RSA, SHA und AES Submodulen)
 * Seite 102 _DPORT_PERIP_CLK_EN_REG (0x0C0)_ (Deaktivieren der meisten anderen Peripherialien)

Für den **ESP8266** habe ich keine derartige Information in der technical reference manual gefunden.
Da werde ich die nächsten Tage ein bisschen google bemühen.

DanielR92, 2022-09-23:
Es wurde damals schon bzgl. Stromaufnahme Tests gemacht, zwar nicht von uns direkt. Kedoch ist die Quelle sehr gut Strukturiert und mit Oszi-Bilder dargestellt.

https://ullisroboterseite.de/esp8266-strom.html

stefan123t, 2022-11-07:
homeautomation hat wohl nachgemessen, wobei der größte Stromfresser im Prinzip die WiFi Option des ESP8266/ESP32 ist. 
Ca. 50-120mA Idle mit Wifi Verbindung und Peaks bis max. 200mA.
Alle noch kürzeren Peaks gleicht ein Elko aus. Ein 2W Netzteil reicht für einen ESP.

Für das nRF24 Modul beim Senden empfehlen und verwenden wir daher i.d.R. einen 47/100uF Elko an den GND und +3.3V Pins 1&2 des NRF24 Moduls.

@thfcm können wir das Issue schließen oder hast Du konkrete Vorschläge wie wir da nennenswert Strom sparen können ?
Man könnte eventuell die Sonnenaufgangs/-untergangszeit nutzen um den ESP in den Deep Sleep bis zum nächsten Morgen zu schicken ?

dj9717, 2022-11-14:
Man könnte eventuell die Sonnenaufgangs/-untergangszeit nutzen um den ESP in den Deep Sleep bis zum nächsten Morgen zu schicken ?
Das hatte ich mir auch bereits überlegt und fände ich ein sehr praktisches Feature.
Ebenso auch eine Möglichkeit den ESP manuell (und via API ) für eine Zeit x in den DeepSleep zu schicken.
Wollte selbst schon Hand anlegen weiß aber mangels KnowHow nicht ob ich es wirklich hinkriegen würde.

dj9717, 2022-11-14:
> Man könnte eventuell die Sonnenaufgangs/-untergangszeit nutzen um den ESP in den Deep Sleep bis zum nächsten Morgen zu schicken ?

Man könnte eventuell die Sonnenaufgangs/-untergangszeit nutzen um den ESP in den Deep Sleep bis zum nächsten Morgen zu schicken ?
Das hatte ich mir auch bereits überlegt und fände ich ein sehr praktisches Feature.
Ebenso auch eine Möglichkeit den ESP manuell (und via API ) für eine Zeit x in den DeepSleep zu schicken.
Wollte selbst schon Hand anlegen weiß aber mangels KnowHow nicht ob ich es wirklich hinkriegen würde.