lumapu/ahoy #268: WR Status per MQTT

Sascha79, 2022-09-12:
Hallo zusammen und danke an alle Beteiligten für die tolle Arbeit. 

Wäre es möglich, den WR Status

- is available and is producing
- is available and is not producing
- is not available

per MQTT zu senden? Oder wenigstens eine entsprechende ID.

c1328, 2022-09-12:
Vielleicht lässt sich das leicht damit kombinieren, dass die Meldungen vom WR auch gleich in "sprechender" Form über MQTT ausgegeben werden, dann spare ich mir das Mapping im Grafana Dashboard (wo ich es für einige Codes inzwischen aber hinterlegt habe).

lumapu, 2022-09-12:
man kann ja beides Senden, eine ID und in ausgeschriebener Form

SandnerSoft, 2022-09-18:
https://github.com/lumapu/ahoy/pull/289
ist da enthalten, bitte prüfen und übernehmen

closed 2022-09-18 by lumapu