lumapu/ahoy #269: Wechsel auf AsyncWebServer

Ziyatoe, 2022-09-12:
man könte damit über WebSerial die console output/input auf HTTP umbiegen. das waere super!
ich habe bereits etwas in die richtung unternommen aber nicht weiter gemacht.
wenn man überall anstatt DEBUG_OUT.print/println, DEBUG_OUT.printf verwendet, kann man diesen printf auf WebSerial umbiegen. 
WebSerial in der basis version hat keinen printf, ich habe den printf für WebSerial hier:
https://github.com/Ziyatoe/WebSerial-printf

stefan123t, 2022-09-12:
Hallo Ziyat T.,
Schön dass Du nun auch auf github aktiv bist. Komme doch mal (wieder?) in den Discord Chat dort besprechen wir solche Sachen auch vorab.
Das og Feature ist praktisch schon fast fertig. Es gibt einen asyncWeb03 branch in dem lumapu bereits den ESPAsyncWebServer und eine WebSerial Console eingebaut hat. Die Tests vor dem Merge sind in PR #107 fast abgeschlossen.

Ziyatoe, 2022-09-12:
ja das ist ne super nachricht! ich hatte eine zwangspause, jetzt langsam werde ich wieder aktiv..

closed 2022-09-18 by lumapu