lumapu/ahoy #278: Setting Power Limit to NoPowerLimit

tbnobody, 2022-09-14:
Ich habe mir gerade den Source angeschaut wie Ahoy das Power Limit zurück setzt. Dabei sind mir speziell folgende 2 Zeilen aufgefallen:

https://github.com/lumapu/ahoy/blob/99c2b9f7a936bad6109a0ab7d8362517dbf15dc0/tools/esp8266/web.cpp#L330
https://github.com/lumapu/ahoy/blob/99c2b9f7a936bad6109a0ab7d8362517dbf15dc0/tools/esp8266/web.cpp#L342

Müsste hier nicht eher `actPwrLimitControl` auf `0xffff` abgefragt werden?

aschiffler, 2022-09-14:
Nein. ActivePwrControl ist nur die Einstellung wie der Wert interpretiert wird. Persistent, Absolut, Relativ usw.

tbnobody, 2022-09-14:
Ja. Oder eben NoPowerLimit was mit dem enum wert 0xffff ankommt

aschiffler, 2022-09-14:
Achso, ja das stimmt. Da kann man die Abfrage verbessern/ändern

tbnobody, 2022-09-14:
Kritisch ist hier wohl Zeile 330 in Verbindung mit Zeile 332
https://github.com/lumapu/ahoy/blob/99c2b9f7a936bad6109a0ab7d8362517dbf15dc0/tools/esp8266/web.cpp#L332

Wenn ein Benutzer aktuell `NoPowerLimit` auswählt, würde `0xffff` an den Inverter gesendet (außer es ist an anderer Stelle nochmal ausgeschlossen).

lumapu, 2022-09-14:
Zeile 330 schließt doch 0xFFFF aus

tbnobody, 2022-09-14:
Wenn man ein Limit von 500 vorgibt und den Modus und NoPowerLimit stellt ist doch das pwrLimit 5000 und pwrLimitControl 0xffff
Dadurch wäre die Bedingung aber wahr und der Code im if block würde ausgeführt

lumapu, 2022-09-14:
Danke für deinen Hinweis, ich habe es jetzt in asyncWeb03 gefixt. Ich denke der Fehler kam nur zum tragen, wenn man das Powerlimit geändert hat, danach aber keinen Neustart des ESPs durchführt (ist aktuell Standard für ESP8266)

DanielR92, 2022-09-15:
Bin mir nicht sicher ob dieses Issue geeignet ist, aber aktuell mit der Version "Git SHA: g4454232 :: 0.5.18" ist es nur hin und wieder möglich die Limitierung über das Setup zu setzen. Da ist anscheinend was kaputt gegangen?

Es betrifft bei mir nur "absolute in Watt persistent".

Kann das jemand bestätigen?

tbnobody, 2022-09-16:
Ohne das ich da jetzt viel drauf rumreiten möchte...
Aber wenn der User jetzt aktuell den Modus auf "NoPowerLimit" stellt, wird das Power Limit ja nicht auf 100% gesetzt...
https://github.com/lumapu/ahoy/blob/4c5a9cb37f023908dfcb431d7cb2e8eee5b3d9eb/tools/esp8266/web.cpp#L243

Imho müsste folgende Zeile
https://github.com/lumapu/ahoy/blob/4c5a9cb37f023908dfcb431d7cb2e8eee5b3d9eb/tools/esp8266/web.cpp#L244
so aussehen:
`if (actPwrLimitControl != NoPowerLimit && actPwrLimit > 0){`

Und folgende Zeile:
https://github.com/lumapu/ahoy/blob/4c5a9cb37f023908dfcb431d7cb2e8eee5b3d9eb/tools/esp8266/web.cpp#L256
so:
`if (actPwrLimitControl == NoPowerLimit ) { // set to 100%`

DanielR92, 2022-09-18:
Ich habe mir das mal angeschaut, beide Variationen entweder vom Originalen oder von tbnobody haben bei mir das problem, das sich keine Änderung bemerkbar macht. 

Im Debug wird Angezeigt:

10:12:12.603 > I: procPyld: cmd:  11
10:12:12.606 > I: procPyld: txid: 0x95
10:12:12.608 > I: Payload (62): 00 01 00 04 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 41 00 01 00 04 00 03 00 0C 00 00 00 07 00 00 0B CD 00 00 00 09 08 E3 13 87 00 00 00 CA 00 00 00 00 00 D8 00 17

------

10:12:55.810 > I: Power limit for inverter 0 set to 40W
10:12:55.814 > I: Power Limit Control Setting 0

Ich kämpfe nähmlich mit dem Problem: 
https://github.com/lumapu/ahoy/issues/287

lumapu, 2022-10-21:
@DanielR92 der issue ist doch obsolet nach eurer letzten Änderung. kann man den schließen?

commit 4c5a9cb3 (2022-09-14): fix #278
closed 2022-10-30 by lumapu