lumapu/ahoy #281: Feature Request | Loginscreen

ambivalent11, 2022-09-15:
Da ich ab und an via DNS auf den AHOY zugreife, wäre es möglich eine Loginmaske zu integrieren?

Ist sicherlich für andere ebenfalls interessant?

hubsi5, 2022-09-15:
Warum nicht per VPN? Ist ja sicherer

stefan123t, 2022-09-15:
Eventuell sinnvoll hierfür dasselbe Password wie beim AP zu verwenden. Dann muss man ggf nur noch ein Input Feld im Setup hinzufügen.

lumapu, 2022-09-15:
ich rate dringend ab so einen Dienst ohne VPN öffentlich ins Netz zu stellen. Auch ein mögliches Passwort schützt hier aus meiner Sicht nicht ausreichend.
Ich finde die Idee generell gut, um im internen Netz die Konfiguration vor dritten zu schützen, wir zB bei einer Fritzbox

lumapu, 2022-10-21:
@DanielR92 wie ist dein Fortschritt? Kann man unterstützen?

stupsi099, 2022-10-30:
ein Passwort wäre eine super Idee. Idealerweise für alles außer LIVE. Ein eigenes wäre sinnvoll da viele ein X-stellige Zahlenkombination beim Wlan haben.

lumapu, 2022-10-31:
natürlich ein eigenes, die Live Übersicht auszunehmen fine ich eine gute Idee, evtl optional

lumapu, 2022-11-02:
@ambivalent11 es gibt eine development Version, die eine Loginmaske hat

closed 2022-11-20 by lumapu