lumapu/ahoy #284: Debug Modus / Serielle Konsole - Inverter #0: HM-1200 (v0) is available and is not producing

J3M3W, 2022-09-16:
FW 0.5.17
HM-1200
D1 Mini Pro

Ich erhalte entweder HM-1200 (v0) is not available and is not producing oder is available and is not producing. Ich sehe aber an der Strom Smart Dose und am blinken das er produziert. Ich wollte jetzt mal schauen bzw. posten was da gesendet wird oder was nicht, allerdings wenn ich den D1 via putty am laptop, seriell, oder mit XTCU connecte bekomme ich nur ab und an ein paar hyroglypen. In den Dokus oder readme habe ich nichts dazu gefunden. Danke

Uptime: 0 Days, 00:04:59; now: 2022-09-16 20:55:32

Statistics:

Receive success: 0
Receive fail: 0
Frames received: 0
Send Cnt: 10

Inverter #0: HM-1200 (v0) is not available and is not producing
Inverter #1: n/a
Inverter #2: n/a
MQTT: not connected

J3M3W, 2022-09-16:
ah es scheint wohl so zu sein falsche oder zu niedrige baud rate im tool = Hieroglyphen

I: enqueuedCmd: 1
I: enqueuedCmd: 5
I: add inverter: HM-1200, SN: 114181043997 
I: connect to network 'SputnikX' ...
...........
I: 
I: RF24 Amp Pwr: RF24_PA_I: HIGH
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_HIGH
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 0 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 15
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Disabled
Primary Mode		= RX
TX address		= 0xdeadbeef01
pipe 0 (closed) bound	= 0xdeadbeef01
pipe 1 ( open ) bound	= 0x1234567801
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: [NTP]: 2022-09-16 21:21:28

I: Requesting Inverter SN 114181043997 
I: Requesting Inverter SN 114181043997 
I: Requesting Inverter SN 114181043997 
I: Requesting Inverter SN 114181043997

lumapu, 2022-09-16:
ja genau, die Baudrate muss 115200 sein. Jetzt ist wahrscheinlich auch die Sonne weg und dadurch keine Kommunikation zum Wechselrichter möglich.
Du könntest im Setup noch serial debug aktiveren

stefan123t, 2022-09-16:
Ändere mal in der `tools/esp8266/include/dbg.h` Datei die folgende Zeile
https://github.com/lumapu/ahoy/blob/99c2b9f7a936bad6109a0ab7d8362517dbf15dc0/tools/esp8266/include/dbg.h#L24
in den Warning Level ab und kompiliere nochmal neu:
```c
#define DEBUG_LEVEL DBG_WARN
```

@lumapu sollten wir das als neuen Default für Release setzen ?
Die Probleme mit zu vielen Ausgaben hatte ich ja mit DEBUG bzw. VERBOSE Level.

J3M3W, 2022-09-17:
Wenn Jemand mal Zeit und Lust hat kann ers ja gerne vorne in der Anleitung mit aufnehmen, weil ich glaube 9600 wird immer gerne per default von allen Programmen genommen.
Gibts eigentlich irgendwo n richtiges help Forum dafür? Weil irgendwie connecten will er nicht. Er war mal  bei einem test einmal connected, allerdings kam da dann nur "available and is not producing" war aber laut smarter dose 100 watt was gerade in Produktion war. 

Ist eigentlich irgendeine Prüfung integriert wo es einen error auswerfen würde wenn mit der sende/empfangseinheit etwas nicht stimmen würde? Wobei was gut war, ich habe mal das 3,3v kabel vom sender abgezogen und es kam ne fehlermeldung auf der webpage

stefan123t, 2022-09-17:
@J3M3W die 115200 bps sind eigentlich der ESP8266/ESP32 Default lediglich der Bootloader schreibt mit 9600 bps seine Meldungen. Wenn man die platformio.ini anschaut ist es dort auch definiert für den Serial monitor. Aber vielleicht sollten wir es noch in die README.md oder die FAQ Wiki Seiten aufnehmen.

@lumapu eigentlich eine gute Frage kann man die NRF24 printPrettyDetails irgendwo im UI als zusätzliche Statusmeldung einbauen ? Vielleicht im Setup bei der Pin Einstellung als Popup auf Knopfdruck ?
Auch eine Ausgabe der gefundenen SSIDs und deren RSSI wurde ja schonmal für die Setup Seite angefragt.

J3M3W, 2022-09-17:
@stefan123t weil mir kam der Gedanke, bei dem Post den Ihr verlinkt habt in Kombination mit meinem Problem.
https://github.com/lumapu/ahoy/issues/230

Denn ich habe hier zwei Module von zwei Läden jeweils in Deutschland, bei denen habe ich schon viel bestellt und ich habe die auch angerufen. Beide haben den Chip mit dem Kreis drauf. Ich kenne das auch noch von einem anderen Projekt mit einem Zusatz Modul von einem Arduino, da wurde auf einem Bauteil eine neue Revision von einem anderen Hersteller verbaut. Specs des Moduls die selben, haben sich aber mit dem Code anders verhalten. Kurz, ich glaube das sind keine Fake Module/Chips das sind entweder von einem anderen Hersteller oder neuere Revs.