lumapu/ahoy #286: System Config Display für ESP32

andreash-esp, 2022-09-17:
System Config zeigt auch beim ESP32 die Bezeichnungen des WEMOS an:

[image]

lumapu, 2022-09-17:
sehr gut es hier festzuhalten, wir werden es implementieren

DanielR92, 2022-09-20:
Ganz kurz, was soll hier bitte geändert werden, dann könnte ich es in die Hand nehmen. :)

lumapu, 2022-09-20:
die Namen und Anzahl der GPIOs sowie der Hinweis no IRQ stimmen nur für ESP8266.
Danke schonmal.

Schorsch980, 2022-09-25:
mein Issue passt hier denke ich thematisch hin: Ist es bei der Gelegenheit möglich, die Liste der auswählbaren IRQs für ESP32 >16 zu erweitern? Problem ist, ich nutze hierfür ein Mini D1 ESP32, dass den gleichen Formfaktor wie D1 Mini (8266) hat - denn ich möchte das gelötete Shield einfach umsetzen. Bei dem Board entspricht D3 = IO17 und da komme ich in der Auswahl nicht dran, weil sie bei IO16 endet. Alle anderen Pins stimmen...

stefan123t, 2022-09-25:
Für den ESP32 build gibt es bis zu GPIO39 wobei GPIO34..39 nur als Input dienen.
Auch einige andere fallen vermutlich aus wegen pullups, etc

Siehe eine Liste hier:

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

Albert2222222, 2022-10-19:
> mein Issue passt hier denke ich thematisch hin: Ist es bei der Gelegenheit möglich, die Liste der auswählbaren IRQs für ESP32 >16 zu erweitern? Problem ist, ich nutze hierfür ein Mini D1 ESP32, dass den gleichen Formfaktor wie D1 Mini (8266) hat - denn ich möchte das gelötete Shield einfach umsetzen. Bei dem Board entspricht D3 = IO17 und da komme ich in der Auswahl nicht dran, weil sie bei IO16 endet. Alle anderen (relevanten) Pins stimmen... [image]

Habe exakt das gleiche Problem, IO17 benötigt für den ESP32 mini!

Albert2222222, 2022-10-20:
Danke daß Du Dich dem angenommen hast @lumapu !

Im akt. Devbuild ist mir nur aufgefallen daß die IOs alle mit "Dxx" bezeichnet sind - müssten die beim ESP32 nicht eigentlich alle "GPIOxx" heißen?

lumapu, 2022-10-20:
das weiß ich auch nicht, aber auf der 'Standard' ESP32 Platine sind die wohl aufgedruckt.
Wobei ich weiß nicht welche die Standard Platine ist.

Albert2222222, 2022-10-20:
[image]

Hier das offizielle Pinout Bild von Espressif, Quelle: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-devkitc.html

GPIOxx sind die offiziellen Bezeichnungen.

lumapu, 2022-10-21:
@andreash-esp bitte mit aktuellem Dev Branch nochmal testen

Albert2222222, 2022-10-23:
Eben mit der 0.5.23 getestet, Zuordnung funktioniert jetzt einwandfrei auf einem ESP32 mini! Danke !!

closed 2022-10-23 by lumapu

mattmeye, 2023-01-19:
@Schorsch980 Darf ich fragen, wie du die anderen belegt hast? D8/GPIO05 ; D4/GPIO 16; D3/GPIO 17 ? Hab ein ähnliches von AZ und es mag einfach nicht sich mit dem RF verbinden.

Schorsch980, 2023-01-20:
@mattmeye nach Anleitung müsste es funktionieren, ich bin damals allerdings zu OpenDTU gewechselt, daher hier keine praktische Erfahrung.