lumapu/ahoy #287: Panic/Crash/Reboot

DanielR92, 2022-09-18:
Steht irgendwo im Code das nach dem x-ten Versuch Daten zu empfangen das ganze Neuzustarten?
AsyncWeb03 - 5.18

Siehe Log:

10:17:33.707 > I: resetPayload: id: 0
10:17:33.710 > I: enqueued cmd failed/timeout
10:17:33.713 > I: Inverter #0 I: no Payload received! (retransmits: 0)
10:17:33.718 > I: Requesting Inverter SN 116174403329
10:17:33.721 > I: enqueuedCmd: 11
10:17:33.722 > I: enqueuedCmd: 5
10:17:33.723 > I: sendTimePacket
10:17:33.725 > I: TX 27B Ch40 | 15 74 40 33 29 78 56 34 12 80 0B 00 63 26 F0 42 00 00 00 1B 00 00 00 00 64 D9 E9
10:17:33.767 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 01 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 90 
10:17:33.782 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 01 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 90 
10:17:33.792 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 01 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 90 
10:17:33.832 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 02 00 00 00 00 00 00 00 00 01 3F 00 01 00 39 00 03 92 
10:17:33.842 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 02 00 00 00 00 00 00 00 00 01 3F 00 01 00 39 00 03 92 
10:17:33.868 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 03 00 B5 00 00 00 07 00 00 0B CD 00 00 00 09 08 DC 3F 
10:17:33.876 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 03 00 B5 00 00 00 07 00 00 0B CD 00 00 00 09 08 DC 3F
10:17:33.886 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 03 00 B5 00 00 00 07 00 00 0B CD 00 00 00 09 08 DC 3F 
10:17:33.897 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 03 00 B5 00 00 00 07 00 00 0B CD 00 00 00 09 08 DC 3F 
10:17:33.928 > I: RX 27B Ch61 | 95 74 40 33 29 74 40 33 29 84 13 85 00 AF 00 C8 00 08 02 92 00 D9 00 1B 6D 37 E0 
10:17:35.258 > I: procPyld: cmd:  11
10:17:35.261 > I: procPyld: txid: 0x95
10:17:35.262 > I: Payload (62): 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 3F 00 01 00 39 00 03 00 B5 00 00 00 07 00 00 0B CD 00 00 00 09 08 DC 13 85 00 AF 00 C8 00 08 02 92 00 D9 00 1B
10:17:37.640 > 
10:17:37.643 > User exception (panic/abort/assert)
10:17:37.645 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
10:17:37.652 >
10:17:37.652 > Unhandled C++ exception: OOM
10:17:37.654 >
10:17:37.655 > >>>stack>>>
10:17:37.656 >
10:17:37.656 > ctx: sys
10:17:37.657 > sp: 3fffedb0 end: 3fffffb0 offset: 0000
10:17:37.660 > 3fffedb0:  3fff9728 3fff1f3c 3fffb0f8 4025d67c  
10:17:37.664 > 3fffedc0:  000000fe 00000000 00000000 00000000
10:17:37.668 > 3fffedd0:  00000000 00000000 00000000 00000001
10:17:37.673 > 3fffede0:  00007fff 00000000 6e978d4f 3fff2388  

UND später danach:

10:19:50.085 > I: sendTimePacket
10:19:50.086 > I: TX 27B Ch75 | 15 74 40 33 29 78 56 34 12 80 0B 00 63 26 F0 CA 00 00 00 1D 00 00 00 00 62 57 EF
10:19:50.135 > I: RX 27B Ch40 | 95 74 40 33 29 74 40 33 29 01 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 90 
10:19:50.170 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 02 00 00 00 00 00 00 00 00 01 3F 00 01 00 46 00 02 EC 
10:19:50.182 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 02 00 00 00 00 00 00 00 00 01 3F 00 01 00 46 00 02 EC 
10:19:50.197 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 02 00 00 00 00 00 00 00 00 01 3F 00 01 00 46 00 02 EC 
10:19:50.221 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 03 00 DD 00 00 00 07 00 00 0B CE 00 00 00 0A 08 E2 69 
10:19:50.231 > I: RX 27B Ch23 | 95 74 40 33 29 74 40 33 29 03 00 DD 00 00 00 07 00 00 0B CE 00 00 00 0A 08 E2 69 
10:19:50.246 > I: RX 27B Ch75 | 95 74 40 33 29 74 40 33 29 03 00 DD 00 00 00 07 00 00 0B CE 00 00 00 0A 08 E2 69 
10:19:50.276 > I: RX 27B Ch3 | 95 74 40 33 29 74 40 33 29 84 13 89 00 D4 00 CA 00 09 02 D5 00 D9 00 1D 6E 5D BC 
10:19:51.597 > I: procPyld: cmd:  11
10:19:51.599 > I: procPyld: txid: 0x95
10:19:51.602 > I: Payload (62): 00 01 00 05 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 3F 00 01 00 46 00 02 00 DD 00 00 00 07 00 00 0B CE 00 00 00 0A 08 E2 13 89 00 D4 00 CA 00 09 02 D5 00 D9 00 1D
10:19:57.274 > 
10:19:57.274 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
10:19:57.280 >
10:19:57.280 > Exception (29):
10:19:57.281 > epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
10:19:57.288 >
10:19:57.288 > >>>stack>>>
10:19:57.289 >
10:19:57.289 > ctx: sys
10:19:57.290 > sp: 3fffec10 end: 3fffffb0 offset: 0190
10:19:57.294 > 3fffeda0:  00000000 3fff0750 00000002 401005c0
10:19:57.299 > 3fffedb0:  4026abbf 00000869 3ffe88c8 4026ab54  
10:19:57.303 > 3fffedc0:  00000002 4026aafb 00000002 40269c50
10:19:57.308 > 3fffedd0:  40269c79 3fffee80 3fff0750 0000001a
10:19:57.312 > 3fffede0:  402676d8 3fffee80 3fff05f8 3fff0014  
10:19:57.318 > 3fffedf0:  3ffede4c 3fffee80 3fffee80 4025e174

DanielR92, 2022-09-18:
Kann das jemand Querprüfen, wenn die Option "Serial Debug" aktiv gesetzt ist?
Wurde das nicht angesprochen das der ESP dann überfordert ist?

Denn Serial Debug hatte ich aktiv, als ich die Abstürze hatte.

stefan123t, 2022-09-18:
Hallo Daniel
Das ist ein 0x15 DevInfo Cmd bzw RealTimeRunData_Debug 0x0B SubCmd

Das Active PowerLimit SubCmd 0x0B beginnt mit 0x51 DevControl. Auch die Antwort beginnt nicht mit 0x95 sondern 0xD1 = 0x51 | 0x80.

stefan123t, 2022-09-18:
Ach ja hier steht es auch:

Unhandled C++ exception: OOM

Also OOM = Out of memory

stefan123t, 2022-09-18:
Ich hatte ähnliche Probleme mit DEBUG_LEVEL = DBG_DEBUG in der include/dbg.h. Mit DBG_INFO klappt es im release Build.
Verwendest Du den release oder den debug Build ? Ich nehme an es ist ein ESP8266 da die ESP32 nicht so schnell in OOM laufen sollten.

DanielR92, 2022-09-19:
Um ehrlich zu sein, nutze ich das auf der Setup-Seite die Einstellung mit dem Debug. 
Also nichts hardcoded... Da geb ich dir recht zwecks Memory, jedoch wie lange geht es gut am ESP32. 

Für mich heißt es, das irgendwo der Speicher voll läauft obwohl man das gar nicht müsste?

stefan123t, 2022-09-19:
@DanielR92 ist der OOM 😱 am ESP32 ?

DanielR92, 2022-09-23:
Ne ich nutze ein ESP8266. Daher wäre beim ESP32 luft nach oben... jedoch je weiter es steigt desto dünenr wird die Luft. 
Daher würde ich gerne dieses Issue solange am leben lassen, bis man sich damit befasst um Präventiv in der Zukunft kein RAM voll zu müllen. :)

DanielR92, 2022-10-13:
Ich close dieses Issue vorerst. 
Da in Zukunft sich bestimmt noch einiges ändern wird.

closed 2022-10-13 by DanielR92