lumapu/ahoy #290: NodeMCU: Reboot-Schleife

reserve85, 2022-09-18:
Hallo,

ich habe meinen NodeMCU ESP32 mit der 220906_ahoy_0.5.17_esp32_5402e9b.bin geflasht, leider startet bei mir die Firmware nicht. 

**Flash-Log:**
Command: esptool.py --port COM3 --baud 230400 --after no_reset write_flash --flash_size detect --flash_mode dio 0x00000 .\220906_ahoy_0.5.17_esp32_5402e9b.bin --erase-all

esptool.py v3.0
Serial port COM3
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:0d:9d:60
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.4s
Compressed 873040 bytes to 550116...
Wrote 873040 bytes (550116 compressed) at 0x00000000 in 24.6 seconds (effective 283.7 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

**Serial Debug Meldung:**

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
...

Kann mir hierzu jemand helfen? Ist die Zieladresse 0x00000 korrekt? Das Funkmodul ist noch nicht angeschlossen, liegt es daran?

reserve85, 2022-09-19:
@stefan123t, leider das gleiche Problem mit 115200 und Erase Flash = 1. Ich werde mich nochmal melden wenn das nRF24L01+ Modul angekommen ist, dachte ich spiele die Firmware schonmal auf und versuche es ins WLAN zu bringen. Scheint wohl so erstmal nicht zu funktionieren. Das verwendete Board ist dieses hier: https://www.berrybase.de/esp32-nodemcu-development-board?c=306

Leider bekomme ich AHoy in VS Code out of the Box erstmal nicht zu compilieren, da muss ich erst nochmal schauen warum.

Aber vorab schonmal Hut ab vor eurer Arbeit und Danke für das tolle Projekt!

reserve85, 2022-09-19:
update: funktioniert nun. Konnte es jetzt in VS Code compilieren und uploaden - das hat geklappt! Nachvollziehen kann ich es allerdings nicht, wieso das vorkompilierte .bin File (bei mir) nicht funktioniert?

stefan123t, 2022-09-20:
Also ich verwende nur PlatformIO daher ist meine Datenbasis natürlich eingeschränkt:
```bash
Creating BIN file ".pio/build/esp8266-release/firmware.bin" using "/home/user/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and ".pio/build/esp8266-release/firmware.elf"
...
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/esp8266-release/firmware.bin
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:3a:e8:42:9d:f0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 616544 bytes to 397755...
...
Wrote 616544 bytes (397755 compressed) at 0x00000000 in 35.1 seconds (effective 140.7 kbit/s)...
Hash of data verified.
```
Es wird auch esptool v3 verwendet aber anscheinend findet er alle notwendigen Optionen "automagisch".

DanielR92, 2022-09-22:
Das beduetet wir können dieses Issue closen?

closed 2022-09-22 by lumapu

bjoerns1983, 2022-09-26:
Also ich habe genau das gleiche Problem konnte es bisher aber nicht lösen. Die Lösung von @stefan123t  will mir aber nicht einleuchten. In dem letzten Beitrag wird ja offensichtlich eins ESP8266 geflashed und kein ESP32. 
Ich habe mehrer Boards auf die gleiche Weise geflashed aber alle zeigen die Serial Debug fehlermeldungen die oben schon aufgeführt sind. Selber kompilieren hat auch nichts geändert.

reserve85, 2022-09-26:
Also ich kann es mit den ganzen Flash Tools auch nicht auf dem ESP zum Laufen bringen. Es funktioniert bei mir nur der Upload mit MS Code, reproduzierbar! Wie hast du es geflasht? Auch mit Ms Code?

bjoerns1983, 2022-09-26:
Habe es mit Visual Studio Code kompiliert und auch flashen lassen. Ergebnis ist immer das gleiche

bjoerns1983, 2022-09-27:
Heute morgen noch mal versucht und nun scheint es mit Visual Studio Code zu funktionieren, woran es auch immer liegt.
Eines kann ich auf jeden Fall festhalten, die vorkompilierte Firmware mit NodeMCU flasher etc. zu flashen funktioniert jedenfalls nicht.

VolkerH, 2022-11-19:
Gleiches Problem mit dem Flashen der `.bin` - Dateien (habe verschiedne Builds probiert) via NodeMCU flasher wie im ersten Post hier. Board ist ein Espressif WROOM dev kit board v1 ESP32 mit 30 Pins.
Vielleich kann man das Issue offen lassen, damit es einfacher gefunden wird.

lumapu, 2022-11-19:
wurde auch Mal über unsere Website geflasht?
https://ahoydtu.de

VolkerH, 2022-11-20:
Nein, habe es mit NodeMCU probiert. Wenn das mit der Webseite die empfohlene Variante ist, waere es ganz gut das etwas prominenter im Github repo zu platzieren. Aber wenn ich das richtig sehe ist das Flashen ueber die Webseite nicht fuer den ESP32 ?
Ich werde es noch mal mit VS Code und PlatformIO versuchen.

lumapu, 2022-11-20:
Das flashen über die Website sollte Standard sein, man braucht nichts besonderes zu installieren oder herunterzuladen.
Es ist sowohl ESP8266 als auch ESP32 möglich.

Wir müssen bzgl. Dokumentation noch viel besser werden, danke für den Hinweis.

VolkerH, 2022-11-21:
Ich habe es gestern abend nochmal mit VSCode/PlatformIO versucht und damit lief es einwandfrei.

schoko24, 2022-11-27:
Auch ich habe mehrere Stunden mit dem Flashen und dem Bootloop verbracht. Ursache ist wirklich das Flashtool. Konnte es eben mit mit einem frischen ESP32 DEVKIT V1 nachtesten: NodeMCU Flasher keinerlei Erfolg, egal ob NFL dran oder nicht dran. Auch der Flashdownloader von Expressif führte zum gleichen Ergebnis.  Nur über die Webseite hat es auf Anhieb funktioniert :-)

stefan123t, 2022-12-02:
@lumapu evtl in der getting_started_with_esp8266.md einfach einen Link auf Deine neue Homepage https://ahoydtu.de/getting_started an prominenter Stelle ganz oben und das Prozedere vielleicht noch einmal kurz unten bei den Optionen ESP Web Tool, VSCode + PlatformIO, ESP Flasher, etc. ?

lumapu, 2022-12-02:
ja gerne, ich habe aktuell keine Ahnung über welchen Weg die Leute zu und finden. Über GitHub, Google, etc.

Wir müssen nur einheitlich lenken, dann haben wir die Mehrheit da wo wir denken dass sie richtig aufgehoben sind

MaxDaNr1, 2023-02-12:
Bei mir ist genau das gleiche Problem. Haben auch einen ESP32. Flashen über NodeMCU Flasher führt immer wieder zum Bootloop. 
Ich hatte vorher OpenDTU drauf. Das hatte bis auf die Verbindung zum WR alles funktioniert. Ich wolllte jetzt mal versuchen ob ich mit Ahoy eine Verbindung zum WR hinbekommen.
Wenn ich den ESP32 über die Homepage flashen will, dann komme ich nicht weiter als bis "Preparing Installation" und nichts weiter passiert.

lumapu, 2023-02-12:
es muss der Button für das Programmieren gedrückt werden.
Andere Möglichkeit, habe ich aber nicht ausprobiert: OpenDTU nochmal installieren und Ahoy per Update Funktion installieren. Es genügt die Firmware bin, die anderen bin's werden nicht benötigt