lumapu/ahoy #301: BROADCAST listening on common pipe ending in 01ULL

stefan123t, 2022-09-22:
Das müsste man testen, aber nur das ich dich korrekt verstehe:

Du möchtestes alle Pipes ungefähr so haben:

Standard:
07:36:17.582 > pipe 0 (closed) bound    = 0xe7e7e7e7e7
07:36:17.585 > pipe 1 ( open ) bound    = 0x1234567801
07:36:17.589 > pipe 2 (closed) bound    = 0xc3
07:36:17.591 > pipe 3 (closed) bound    = 0xc4
07:36:17.593 > pipe 4 (closed) bound    = 0xc5
07:36:17.596 > pipe 5 (closed) bound    = 0xc6

Broadcast:
07:36:17.582 > pipe 0 (closed) bound    = 0xe7e7e7e7e7
07:36:17.585 > pipe 1 ( open ) bound    = 0x1234567801
07:36:17.589 > pipe 2 (open ) bound    = 0xc01
07:36:17.591 > pipe 3 (open ) bound    = 0xc01
07:36:17.593 > pipe 4 (open ) bound    = 0xc01
07:36:17.596 > pipe 5 (open ) bound    = 0xc01

__Originally posted by @DanielR92 in https://github.com/lumapu/ahoy/issues/292#issuecomment-1254552180__

stefan123t, 2022-09-22:
Entweder so oder man kann sogar die pipe1 auf 0x01 setzen. Ich kenne mich da auch (noch) nicht aus.

DanielR92, 2022-09-22:
Nur das die Erfahrung weitergeteilt wird:

Ich habe nun jede Pipe (außer 0) auf 0x01ULL geädnert. Die normale Sequenz zur Datenerfassung des WR wird dadurch nicht mehr gewährleistet. Ich behaupte das die ID zu kurz ist und das gebaute Paket am WR ignoriert wird.

Aus der NRF Doku geht heraus das ReadingPipes bis zu 6 Sück gleichzeitig geöffnet werden können:
https://maniacbug.github.io/RF24/classRF24.html#aa7e8523f86f9f8f20c274e0c89a5fd45

Bei WritingPipes jedoch immer nur eins aktiv genutzt werden kann:
https://maniacbug.github.io/RF24/classRF24.html#a50c8e68ee840e1860a31dbdc83afbd77