lumapu/ahoy #302: Inverter ein-/ausschaltern via MQTT

ntfrnd, 2022-09-23:
Hallo,

im Manual gibt es ein Beispiel um mittels MQTT den Inverter ein- bzw. auszuschalten.

"So as example sending any payload on inverter/devcontrol/0/1 will switch off the inverter."

Aufgrund der Liste:
[image]

und des Beispiels gehe ich davon aus, dass mittels MQTT-Message an 
inverter/devcontrol/0/1 der Inverter AUS
inverter/devcontrol/0/0 der Inverter EIN
geschaltet werden kann.

Ich habe -analog zum ActivePowerControl- die beiden Variablen händisch angelegt. 
Sende ich nun z.B. 
    `setState("mqtt.0.inverter.devcontrol.0.1", '1')`
sehe ich dies zwar in der MQTT-Objektansicht und auch im MQTT-Explorer (zur Kontrolle), aber den Inverter interessiert das irgendwie nicht.

Hab ich einen Denkfehler oder irgendwas vergessen?

[image]

stefan123t, 2022-09-23:
Steht das evtl noch nicht im User Manual ?

https://github.com/lumapu/ahoy/blob/main/tools/esp8266/User_Manual.md

ntfrnd, 2022-09-23:
Dort habe ich es gelesen, aber entweder habe ich es nicht verstanden oder es funktioniert so nicht. Deshalb die Nachfrage...

lumapu, 2022-09-23:
welche Version ist im Einsatz? Es kann sein, das es in der development Version nicht funktioniert

ntfrnd, 2022-09-23:
0.5.17 (main) am 09.09. kompiliert und seitdem im Einsatz.

lumapu, 2022-10-21:
@DanielR92 habt ihr buttons mal getestet unter der Webserial?
Sollte sich prinzipiell funktionieren.

stefan123t, 2022-11-07:
@ntfrnd ist das Feature über MQTT noch notwendig oder reicht die Option über die JSON API `/api1` ?

ntfrnd, 2022-11-08:
Ich arbeite nur mit den MQTT-Objekten, von daher würde ich es mir wünschen 😀

DanielR92, 2022-11-17:
Hallo zusammen,

folgendes habe ich getestet (ESP32, v0.5.39):
@lumapu: Serial Console: Turn On + Turn Off => Check, funktioniert.

stefan123t, 2022-11-18:
@DanielR92 das war jetzt der Test dass die JSON API funktioniert. Danke!
Jetzt müsste man sich ansehen ob/wie der MQTT Callback funtkioniert. Irgendwie muss er ja auch das Topic vom MQTT Broker lesen, dann das entsprechende Kommando ansetzen und vermutlich das MQTT Topic zurück setzen ?
Das hatte mW seinerzeit noch @aschiffler implementiert. Vielleicht kann er ja nochmal in den Code schauen ?

lumapu, 2022-11-21:
am besten testen und gleich gut dokumentieren.

lumapu, 2022-11-27:
mit der aktuellen Dev Version sollte das jetzt funktionieren

stefan123t, 2022-11-30:
@ntfrnd kannst Du das neue Feature mit der Version 0.5.44 testen ?

closed 2023-03-08 by lumapu