lumapu/ahoy #304: Implement MainCmd REQ_ARW_DAT_ALL 0x15 SubCmd GetLossRate 0x15

stefan123t, 2022-09-23:
Adding as separate festure request to keep track:

2. the RSSI for the individual inverters can not be given. Though there are some similar values:
* start a AhoyDTU statistic counting the packets received per inverter and each of the five TX/RX channel combinations. This could greatly enhance our channel hopping algorithm for both TX and RX channels !
* there is a DevInfo Sub command to query the inverter for its RX/TX statistics: using MainCmd REQ_ARW_DAT_ALL 0x15 SubCmd GetLossRate 0x15. See #145 and Wiki https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#getlossrate--0x15
* nRF24 can yield a value for each target being above/below the RSSI threshold of 60/64dBm (value depending on source)

__Originally posted by @stefan123t in https://github.com/lumapu/ahoy/issues/260#issuecomment-1240747110__

lumapu, 2023-09-16:
@oberfritze, @rejoe2 kennt ihr das Kommando? 

hier ein kleine Python Script, mit dem du zumindest die Payload im Seriellen Fenster bekommen solltest:

```python
import json
import requests

espUrl="http://192.168.1.41/api"

def getApi(iv, cmd):
    r = requests.post(espUrl + "/ctrl", json={
        "id": iv,
        "cmd": "dev",
        "val": cmd
    })
    if r.status_code == 200:
        obj = r.json()
        print(obj)
    else:
        print(f"error http: {r.status_code}")

getApi(0, 15)
```

muss ich bei Tageslicht gleich mal ausprobieren was da kommt.

lumapu, 2023-12-28:
getllossrate ist in aktuellen Fassungen integriert, zB. 0.8.29

stefan123t, 2023-12-28:
Ja, prima! Danke der Feature Request kann mE geschlossen werden oder fehlt dazu noch ein UI widget ? 
Ich habe es bisher nur in Logs von Knickohr und anderen gesehen/beobachtet.

rejoe2, 2023-12-28:
> Ja, prima! Danke der Feature Request kann mE geschlossen werden oder fehlt dazu noch ein UI widget ? Ich habe es bisher nur in Logs von Knickohr und anderen gesehen/beobachtet.

Falls (!) man da was machen wollte, müßte man die (errechneten) Daten persistent verfügbar machen, im Moment werden die nur "on the fly" errechnet und in die serielle Konsole ausgegeben.
Wenn, würde ich dazu neigen, das in die tx/rx-Statistik/Inverter mit einzubauen. Ist im Prinzip dasselbe Thema, nur eben bezogen auf einen kürzeren Zeitraum...

closed 2024-02-17 by lumapu