lumapu/ahoy #305: Use Preferences.h instead of EEPROM.h for ESP32

stefan123t, 2022-09-23:
As discussed on Discord using the ESP8266 library EEPROM.h does not compute on ESP32. Use Preferences.h instead.

https://github.com/lumapu/ahoy/blob/f3e1bf2faed297b434418e3ae9c96a8a28140930/tools/esp8266/eep.h#L10

Nini12345678901, 2022-09-25:
@stefan123t Hi I'm interested in making a contribution as a good first issue. Could I receive more information as to why the the #include statement needs to be changed to `Preferences.h`? And is that the main change that needs to be done for fixing this issue?

stefan123t, 2022-09-25:
I dunno whether that is ALL that is required to make it run on ESP32 but currently the build fails afaik.

Nini12345678901, 2022-09-26:
@stefan123t Ok could I receive guidance as to where I can access files that are included in `eep.h` such as `EEPROM.h` and potentially `Preferences.h`? Also I would appreciate if I could get more background information or if there is a place I can get more information on these specific files that could help me in navigating the issue such as why `Preferences.h` should be the replacement for the include statement

stefan123t, 2022-09-26:
ESP8266 board files include the EEPROM.h and ESP32 board files include the Preferences.h

It is simply for ESP32 we need to include a different lib.

You can either check the path in your build logs (eg in VScode with PlatformIO plugin) or use the Ctrl+RMB on the references to open them if you want to dig deeper.

Both are standard Arduino-esp8266 or Arduino-esp32 board files as available on github from espressif's igor et al.

Nini12345678901, 2022-09-27:
Ok thanks I will look into this and get to debugging the issue on my IDE

stefan123t, 2022-10-03:
@Nini12345678901 thanks for creating PR #323 .
Please check how we can merge the proposed changes for ESP32 platform so that the ESP8266 platform build still compiles successfully.

lumapu, 2022-11-27:
don't need this any more, because we switched to littlefs

closed 2022-11-27 by lumapu