lumapu/ahoy #306: MQTT Message fehlerhaft, bei mehr als 1 WR

SandnerSoft, 2022-09-24:
Ich habe 3 WR laufen. 
Bei den MQTT Messages YieldDay / YieldTotal wechseln die Werte alle WR immer durch, sieht man hier gut im Bild.
[image]

Außerdem stimmen dadurch die Werte im Topic "total" auch nicht:
[image]

Nutze den development03 Branch und der Fehler ist am ESP32 genauso wie am ESP8266

stefan123t, 2022-09-26:
@SandnerSoft machst Du selbst einen PR für die Korrektur ?
Ich vermute es liegt daran, dass er aktuell einige Daten unter das globale Topic schreibt die eigentlich pro Inverter sind.

SandnerSoft, 2022-09-26:
> @SandnerSoft machst Du selbst einen PR für die Korrektur ? Ich vermute es liegt daran, dass er aktuell einige Daten unter das globale Topic schreibt die eigentlich pro Inverter sind.

ich schau mir das nochmal an, aber bisher konnte ich den Fehler nicht lokalisieren.
Ich probiere mal alle globalen Topics auszublenden.

stefan123t, 2022-10-03:
Man bräuchte also für die Total Werte ein Valid Flag pro WR. 
Diese Valid Flags werden dann beim Berechnen des Total Wertes vorher geprüft und anschließend zurückgesetzt ?

SandnerSoft, 2022-10-04:
Ja genau! 
Irgendwie sowas muss an dieser Stelle eingebaut werden, ansonsten hat man nie alle WR zusammen in den Total Werten.

Chrischi-, 2022-10-09:
Gibt es hierzu schon einen PR?

lumapu, 2022-10-11:
@stefan123t: warum zurücksetzen? Ich glaube sobald man weiß, dass die Payload korrekt ist kann man doch immer wieder die Summe schicken. Sobald mal ein WR keinen Empfang hat muss man das valid-Flag wieder zurücksetzen.

lumapu, 2022-10-31:
@SandnerSoft bitte nochmal mit dem aktuellen Release 0.5.28 testen, MQTT würde in diesem Bereich umgebaut / gefixt

SandnerSoft, 2022-10-31:
ja passt jetzt, vielen Dank

closed 2022-10-31 by lumapu