lumapu/ahoy #307: PowerLimit in setup UI does work in 0.5.14 only

stefan123t, 2022-09-24:
According to out #support channel users have to downgrade via OTA to version 0.5.14 in order to set the powerlimit back to 100% permanent.
The 0.5.14 bin file can be found here: 
https://filehorst.de/d/euEepgdC

stefan123t, 2022-09-24:
To reset flash the bin file to your ESP via OTA and select the following:

ActivePowerLimit: 100
ActivePowerLimit ControlType: relative in percent persistent
[x] reboot after save
Save

lumapu, 2022-11-15:
@stefan123t das geht doch mit den aktuellen development vertreiben wieder, ist jetzt unter Serial Console

stefan123t, 2022-11-17:
Ja aber in unserem Main release ist es mW noch broken oder hast Du schon wieder mal ein Release gebaut bei dem das behoben ist ?
Die Development Version zu verwenden ist allerdings eine Option anstelle des Downgrades.

closed 2022-11-21 by lumapu