lumapu/ahoy #314: Statische IP

vobo-1969, 2022-09-26:
Es wäre super, wenn man eine statische IP einstellen könnte.
Könnte das vielleicht bei einer der nächsten Versionen mit eingebaut werden ?

Viele Grüße
Volker

DanielR92, 2022-09-26:
Gute Idee, müssten man sich anschauen. Sollte aber nichts großes sein.

homeautomation2022, 2022-09-27:
Also ich habs bei mir in [ahoywifi.cpp](ahoywifi::setupStation) hardcoded so drin:
`    IPAddress IP(192, 168, x, x);
    IPAddress Subnetzmaske(255, 255, 255, 0);
    IPAddress Gateway(192, 168, x, x);
    IPAddress DNS(192, 168, x, x);

    WiFi.persistent(false); // verhindert das neuschreiben des Flash
    WiFi.setOutputPower(20.5); // max Tx Power
    WiFi.mode(WIFI_STA);
    WiFi.config(IP, Gateway, Subnetzmaske, DNS, IPAddress(1, 1, 1, 1)); // DNS 2 = fix 1.1.1.1
    WiFi.begin(mSysCfg->stationSsid, mSysCfg->stationPwd);`
Bei den x natürlich die eigenen Daten eintragen und IP möglichst außerhalb des DHCP Bereiches, um eine Doppeltvergabe zu vermeiden.

stefan123t, 2022-10-03:
@homeautomation2022 / @vobo-1969 kann man die feste IP nicht einfach anstelle von AHOY-DTU im Setup unter Device Name als statische IP 192.168.178.XXX eingeben ?
Das wird doch mW immer per WiFi.setHostname() gesetzt und die Methode ist mW auch für eine numerische IP Adresse überladen.

homeautomation2022, 2022-10-04:
Wenn man es sehr einfach halten möchte, müsste man im Setup 2 Felder hinzufügen:
IP: XXX.XXX.XXX.XXX
Gateway/DNS: XXX.XXX.XXX.XXX (hier würden auch nur der letzte Block reichen, da ja Subnetz fest ist, also könnte man anhand der IP automatisch die ersten 3 Blöcke übernehmen)
q'n'd wäre nur die IP eingeben und Gateway/DNS fest die 1 hinten, haben aber auch nicht alle Haushalte so, nur 99,99%.

"Das wird doch mW immer per WiFi.setHostname() gesetzt und die Methode ist mW auch für eine numerische IP Adresse überladen." -> verstehe ich nicht
Hostname setzt nur den DTU Namen für das Netzwerk. Hier war nur mein Einwand, das dieses vor WiFi begin eigentlich aufgerufen werden soll.

fila612, 2022-10-05:
> ich vergebe auch gerne statische IPs im Gerät, aber in dem Fall habe ich es über die Fritz!Box statisch gemacht.

das ist meiner Meinung nach auch die viel bessere Lösung! mach ich genauso :)

homeautomation2022, 2022-10-05:
Nicht jeder hat eine Fritte...
Außerdem ist es genau genommen auch keine statische IP, sondern einfach nur im DHCP fest zugeordnete IP zu der MAC.

CBRocker, 2022-10-05:
und da Ahoy-DTU jeden Tag mit dem Router verbunden ist, sollte die IP wohl eh gleich bleiben (im Normalfall :-) ).
Auf jeden Fall finde ich Ahoy super gelöst, macht viel Spaß damit. Es läuft bei mir super.

deep470, 2022-10-05:
Statische IP würde ich auch begrüßen. Habe hier zwei Netze und würde den Ahoy auf Dauer gerne im DHCP-freien Netz haben. Mein Gateway endet auch nicht auf .1. Feste DHCP-Zuweisung im DHCP-Server ist immer ein Problem wenn der Router ersetzt werden muss oder wenn Gäste mal versehentlich einen Router falsch anschließen und ein zweiter DHCP-Server Adressen vergibt. Meiner Meinung nach gibt es gute Gründe, eine statische IP als Konfigurationsoption zu haben. Ich sehe aber auch ein, dass dieser Feature-Request nicht die höchste Priorität hat. Und ja, danke - Ahoy läuft echt super!

lumapu, 2022-10-05:
> Wenn man es sehr einfach halten möchte, müsste man im Setup 1 oder 2 Felder hinzufügen:
> IP: XXX.XXX.XXX.XXX (wenn leer = DHCP)
> Gateway/DNS: XXX.XXX.XXX.XXX (hier würde auch nur der letzte Block reichen, da ja Subnetz fest ist, also könnte man anhand der IP automatisch die ersten 3 Blöcke übernehmen)
> q'n'd wäre nur die IP eingeben und Gateway/DNS fest die 1 hinten, haben aber auch nicht alle Haushalte so, nur 99,99%.

Man braucht zwei Felder, da ein Netz auch größer als 8 Bit sein kann. Ich habe zum Beispiel ein 10.20.0.0/21 Netz der DNS ist auf 10.20.0.5 und der DHCP hat 10.20.2.0/24
Ich habe es bei mir lieber zentral verwaltet, aber sehe hier schon die Möglichkeit es zu implementieren.
Die Felder müssten dann so sein:
- IP: xxx.xxx.xxx.xxx/xx
- DNS: xxx.xxx.xxx.xxx

In der Web UI kann man ja DHCP anzeigen, wenn man eine leere IP erkennt.

jogibaer65, 2022-10-06:
Ich habe es so gelöst:

in config_override.h
//IP-Adresse vorgeben
#undef WIFI_IP
#define WIFI_IP     IPAddress(192, 168, xxx, xxx)
#undef WIFI_GW
#define WIFI_GW     IPAddress(192, 168, xxx, xxx)
#undef WIFI_DNS
#define WIFI_DNS    IPAddress(192, 168, xxx, xxx)
#undef WIFI_MASK
#define WIFI_MASK   IPAddress(255, 255, 255, 0)

und in ahoywifi.cpp
WiFi.config(WIFI_IP, WIFI_GW, WIFI_MASK, WIFI_DNS);

Könnte man nicht die Zeile WiFI.config.... aufnehmen? Dann blieben die Einstellungen für die IP-Adresse erhalten.

stefan123t, 2022-10-06:
@lumapu ja Mask hast Du an Deiner IP mit angehängt aber das Gateway fehlt auf jeden Fall noch als extra Feld. Das ist ja nicht zwingend auf dem DNS zu finden.
Am Einfachsten / Besten also alle vier Felder wie es @jogibaer65 beschrieben hat oder alternativ per DHCP.

lumapu, 2022-11-06:
implemented, please test

closed 2022-11-06 by lumapu