lumapu/ahoy #315: ESP32 Compile Error

drdigital-ips, 2022-09-27:
Ich bekomme bei Build esp32-wroom32-release nachfolgende Fehlermeldungen:
`[{
	"resource": "/VSCODE/ahoy/tools/esp8266/hmRadio.h",
	"owner": "cpp",
	"severity": 8,
	"message": "there are no arguments to 'DHEX' that depend on a template parameter, so a declaration of 'DHEX' must be available [-fpermissive]",
	"startLineNumber": 258,
	"startColumn": 17,
	"endLineNumber": 258,
	"endColumn": 17
}]

[{
	"resource": "/VSCODE/ahoy/tools/esp8266/hmRadio.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'DHEX' was not declared in this scope",
	"startLineNumber": 258,
	"startColumn": 21,
	"endLineNumber": 258,
	"endColumn": 21
}]

[{
	"resource": "/VSCODE/ahoy/tools/esp8266/hmInverter.h",
	"owner": "C/C++",
	"code": "457",
	"severity": 8,
	"message": "\"\"calcFunctions\"\" ist keine Funktion oder kein statischer Datenmember.",
	"source": "C/C++",
	"startLineNumber": 87,
	"startColumn": 21,
	"endLineNumber": 87,
	"endColumn": 34
}]

[{
	"resource": "/VSCODE/ahoy/tools/esp8266/hmInverter.h",
	"owner": "cpp",
	"severity": 4,
	"message": "variable templates only available with -std=c++14 or -std=gnu++14",
	"startLineNumber": 87,
	"startColumn": 21,
	"endLineNumber": 87,
	"endColumn": 21
}]`
Mache ich etwas falsch?
Oder ist das ein Fehler im code?

SandnerSoft, 2022-09-27:
Hallo, 

du musst dein VS Code -> Plattform IO -> Plattforms ESP32 auf den neusten Stand bringen.
Version > 5.1.0 ist erforderlich!

[image]

drdigital-ips, 2022-09-27:
ok, Danke das wars.
Nun klappt es.

closed 2022-09-27 by drdigital-ips