lumapu/ahoy #320: 1-Bit RSSI (RPD) or “poor man's RSSI” for NRF24L01+

Chrischi-, 2022-10-02:
As you know, the NRF24L01+ does not have an RSSI module like the nRF5 series. 
However, the NRF24L01+ has a type of 1-bit RSSI.

Received Power Detector (see screenshot)
[image]

Wouldn't it make sense to display at least this information in the web interface so that you know that you have 
an RSSI of above or below -64dBm? This could certainly be helpful when choosing a location for ahoy.

stefan123t, 2022-11-22:
M.E. würde es mehr Sinn machen das [GetLossRate](https://github.com/lumapu/ahoy/wiki/Protocol#getlossrate--0x15) 0x15 Kommando zu implementieren.

stefan123t, 2022-11-22:
Siehe #304

closed 2022-11-22 by stefan123t