lumapu/ahoy #329: ntp server in 5.18 not working, causing not to connect to hoymiles

bwired-nl, 2022-10-06:
I updated to 5.18 because 5.17 was very instable.
5.18 handles much better, only problem is that i cannot get NTP to work
The date keeps on ESP-Time: 1.1.1970, 00:00:00
THIS was working good on 5.17
I tried serveral ntp servers , windows, google, diff pool.ntp.org, local ntp server of my synology
rebooted several times etc etc
but I cannot get it to work, beacuse the NTP is not working I cant connect to the Hoymiles also
MQtt is updating every 30 secs version only

stefan123t, 2022-10-06:
NTP is only queried on Start of ESP and every 12 hours. I would also vote to update it via Button in Setup UI or at least after saving. Did you restart your ESP after saving ?

bwired-nl, 2022-10-06:
OK got it 🙂  i see ntp is loading!! 🙂  its something else, looks like it reboots . see below log
hope you can help me!
thanks

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
D: start: 0, length: 113
D: CRC RD: e477 CRC CALC: e477
D: start: 115, length: 562
D: CRC RD: 3505 CRC CALC: 3505
I: add inverter: hoymiles1, SN: 116181852181, Power Limit: 1600 in %
I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
fpm close 1 
mode : sta(b4:e6:2d:6d:d4:1b)
add if0
wifi evt: 8
wifi evt: 2
I: connect to network 'Bwired' ...
........scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with Bwired, channel 6
dhcp client start...
wifi evt: 0
ip:10.0.0.199,mask:255.255.255.0,gw:10.0.0.1
wifi evt: 3
..
I: 

----------------------------------------
I: Welcome to AHOY!
I: 
point your browser to http://10.0.0.199/
I: to configure your device
I: ----------------------------------------
D: mqtt.h:reconnect
D: MQTT mClient->_state -1
D: WIFI mEspClient.status 0
I: RF24 Amp Pwr: RF24_PA_I: HIGH
I: Radio Config:
SPI Frequency        = 1 Mhz
Channel            = 3 (~ 2403 MHz)
Model            = nRF24L01+
RF Data Rate        = 250 KBPS
RF Power Amplifier    = PA_HIGH
RF Low Noise Amplifier    = Enabled
CRC Length        = 16 bits
Address Length        = 5 bytes
Static Payload Length    = 32 bytes
Auto Retry Delay    = 250 microseconds
Auto Retry Attempts    = 0 maximum
Packets lost on
    current channel    = 0
Retry attempts made for
    last transmission    = 0
Multicast        = Disabled
Custom ACK Payload    = Disabled
Dynamic Payloads    = Enabled
Auto Acknowledgment    = Disabled
Primary Mode        = RX
TX address        = 0xe7e7e7e7e7
pipe 0 (closed) bound    = 0xe7e7e7e7e7
pipe 1 ( open ) bound    = 0x1234567801
pipe 2 (closed) bound    = 0xc3
pipe 3 (closed) bound    = 0xc4
pipe 4 (closed) bound    = 0xc5
pipe 5 (closed) bound    = 0xc6
[hostByName] request IP for: 0.nl.pool.ntp.org
[hostByName] Host: 0.nl.pool.ntp.org IP: 5.200.6.34
:urn 48
:urd 48, 48, 0
I: [NTP]: 2022-10-06 23:00:31
D: mqtt.h:reconnect
D: MQTT mClient->_state -1
D: WIFI mEspClient.status 0
[hostByName] Host: 10.0.0.221 is a IP!
:ref 1
:wr 22 0
:wrc 22 22 0
:ack 22
:rn 4
:c0 1, 4
I: subscribe to hoymiles/devcontrol/#
:wr 28 0
:wrc 28 28 0
I: resetPayload: id: 0
D: processPayload - set complete
:wr 68 0
:wrc 68 68 0
:wr 33 0
:wrc 33 33 0

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

DanielR92, 2022-10-11:
In my case it work normaly. Can somebody check that too?

stefan123t, 2022-10-11:
@bwired-nl you are running a debug build and not the release build.
Could you flash / revert your ESP to use the release build.
The debug build has several debug flags enabled that show a lot of events from the ESP Arduino core libraries, e.g. the following WiFi events. Without these events being logged memory and IO resources of the ESP are lower and the release build should run more stable.
```
wifi evt: 0
``` 
Also your nRF24 Module is set to PA_HIGH, is this necessary because your module does not have an external antenna ?
And do you have an ~100uF electrolytic capacitor connected to the VCC 3.3V and GND Pins 1 & 2 of your NRF24 Module to stabilize its power source ?

lumapu, 2022-11-06:
got now feedback, seems to be solved.

closed 2022-11-06 by lumapu